pothosware/SoapySDR #161: SoapySDRUtil --find No modules found

lvywn, 2018-03-31:
When I type SoapySDRUtil --info in Terminal I got this
```
Lib Version: v0.7.0-g824d83e8
API Version: v0.7.0
ABI Version: v0.7
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.7
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, `S8]`
 -    U8 -> [F32, S16, S8]
```
It seems there is something missing. In usr/local/lib/SoapySDR,there is only modules0.6.
And I create a folder names module 0.7 and copy libLMS7Support.so to it.  it turns
```
Lib Version: v0.7.0-g824d83e8
API Version: v0.7.0
ABI Version: v0.7
Install root: /usr/local
Search path: /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/local/lib/SoapySDR/modules0.7/libLMS7Support.so
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

```
When I type ScoapySDRUtil --find
```
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules0.7/libLMS7Support.so)
  lime failed ABI check: Library ABI=0.7, Module ABI=0.6
No devices found!
```
I think what SoapySDRUtil need is libLMS7Support.so whose ABI = 0.7, So what should I do?Thanks.

guruofquality, 2018-03-31:
Careful with copying files around. You should rebuild limesuite after upgrading soapysdr.

closed 2018-04-09 by guruofquality

gvanem, 2018-06-28:
`SoapySDRUtil --info` should IMHO report missing paths (and empty directories) specified in `SOAPY_SDR_PLUGIN_PATH`.