pothosware/SoapySDR #164: Soapy-LimeSDR fails upon setting two channels

myotis6, 2018-05-12:
Soapy source - LimeSDR malfunctions upon setting the channels number to 2. The source runs a LimeSDR-USB card properly when set to 1 channel. The exact same topology then throws errors trying to run two such as:

[14:55:32.053000] SoapyBlock: call setGain threw: Exception: vector::_M_range_check: __n (which is 1) >= this->size() (which is 1)
[14:55:32.053000] SoapyBlock: call setBandwidth threw: Exception: vector::_M_range_check: __n (which is 1) >= this->size() (which is 1)
[14:55:32.053000] SoapyBlock: call setDCOffsetMode threw: Exception: vector::_M_range_check: __n (which is 1) >= this->size() (which is 1)

Different host computers render different sets of errors. One persistently reports resource unavailable, another fails to set synthesizers and the analog filter. Oddly, setting channels back to 1 does not bring back operation. Resetting the SDR and computer does not help. Some combination of resetting everything and creating a new blank graph is the only way I have recovered so far.

LimeSuiteGUI using both RX channels and SDRAngel continue to operate the LimeSDR after Pothos gets into this persistent error state.

In any case, I need two channel receive operation and am asking for help overcoming this problem.

I am running Ubuntu 16.04LTS, Pothos 0.6.0 ppa-1 xenial, Limesuite 18.04.1, LimeSDR hardware 4; protocol 1; firmware 4; gateware 2.16.

TIA

guruofquality, 2018-05-13:
can you share the minimal design that duplicates the issue (saved .pothos file) to make things easy to replicate the issue?

> I am running Ubuntu 16.04LTS, Pothos 0.6.0 ppa-1 xenial, Limesuite 18.04.1, LimeSDR hardware 4; protocol 1; firmware 4; gateware 2.16

PS thanks for all of the version numbers

myotis6, 2018-05-13:
Here's the 'simplest' test that fails.
Part of the problem is that I may be misunderstanding the params. For Channels, is it a list or a count of channels? [1] works, but [0], [2], [0,1], [1,2] fail. When does one use [], {}, "", or just bare numbers?
Repeated reloading eventually leads to "resource busy" errors even though Pothos is all that is running the LimeSDR.
How does one dig deeper?
"issues" here does not allow .pothos files.
Thanks
[image]

myotis6, 2018-05-14:
Error example topology/graph passed as .zip.
[SimpletestSoapy.zip](https://github.com/pothosware/SoapySDR/files/2002229/SimpletestSoapy.zip)

If someone can share the details of setting up a working two channel Soapy source - LimeSDR-USB, that may help.

TIA

Heshyo, 2018-10-26:
The sample rate and/or bandwidth of you topology is too low. If I use 5e6 for sample rate and bandwidth, it works.

guruofquality, 2018-10-27:
In the future let the LimeSuite folks know if there is an issue, the driver code is actually managed there and if it has a bug, their devs need to know: https://github.com/myriadrf/LimeSuite

I know they had a few recent bug fixes for dual channel that should be in release now.

> The sample rate and/or bandwidth of you topology is too low. If I use 5e6 for sample rate and bandwidth, it works.

Thanks!

closed 2018-10-27 by guruofquality