pothosware/SoapySDR #166: Support the USRP devices

ghulands, 2018-05-18:
It would be great if you can also support the UHD/USRP driver.

guruofquality, 2018-05-18:
Took all night, but I got something together real quick, can you try it out? https://github.com/pothosware/SoapyUHD/wiki

ghulands, 2018-05-18:
I'll give it a go tonight and report back.

guruofquality, 2018-05-22:
Thanks all

closed 2018-05-22 by guruofquality

ghulands, 2018-05-23:
I was unable to test it in CubicSDR due to this bug https://github.com/cjcliffe/CubicSDR/issues/569

ghulands, 2018-05-28:
When trying to get it to use the USRP, I just spewed tons off errors to the console.
See https://gist.github.com/ghulands/4299057ff21042db0438c12364537adb
Any ideas?

guruofquality, 2018-05-28:
Looks like reads coming from the usrp are corrupt: `recv_packet_demuxer_3000.hpp:77] [STREAMER] recv packet demuxer unexpected sid 0xffff00` check your drivers, firmware version, see if standalone app works

ghulands, 2018-05-28:
```
$ uhd_rx_nogui -f 104100000 -m FM -A RX2 -g 16 worked fine.

$ SoapySDRUtil --probe="driver=uhd,type=b200"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=uhd,type=b200
[INFO] [UHD] linux; GNU C++ version 7.3.0; Boost_106501; UHD_3.11.1.0-release
[INFO] [b200_impl.cpp:382] [B200] Detected Device: B210
[INFO] [B200] Detected Device: B210
[INFO] [b200_impl.cpp:429] [B200] Operating over USB 3.
[INFO] [B200] Operating over USB 3.
[INFO] [b200_impl.cpp:544] [B200] Initialize CODEC control...
[INFO] [B200] Initialize CODEC control...
[INFO] [b200_impl.cpp:601] [B200] Initialize Radio control...
[INFO] [B200] Initialize Radio control...
[INFO] [b200_impl.cpp:890] [B200] Performing register loopback test... 
[INFO] [B200] Performing register loopback test... 
[INFO] [b200_impl.cpp:899] [B200] Register loopback test passed
[INFO] [B200] Register loopback test passed
[INFO] [b200_impl.cpp:890] [B200] Performing register loopback test... 
[INFO] [B200] Performing register loopback test... 
[INFO] [b200_impl.cpp:899] [B200] Register loopback test passed
[INFO] [B200] Register loopback test passed
[INFO] [b200_impl.cpp:709] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [b200_impl.cpp:935] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [b200_impl.cpp:946] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [B200] Actually got clock rate 16.000000 MHz.

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=b200
  hardware=B210
  fpga_version=14.0
  fw_version=8.0
  mboard_id=B210
  mboard_name=
  mboard_serial=F4FFFF
  rx0_antenna=RX2
  rx0_id=Unknown (0xffff)
  rx0_serial=
  rx0_subdev_name=FE-RX2
  rx0_subdev_spec=A:A A:B
  rx1_antenna=RX2
  rx1_id=Unknown (0xffff)
  rx1_serial=
  rx1_subdev_name=FE-RX1
  rx1_subdev_spec=A:A A:B
  tx0_antenna=TX/RX
  tx0_id=Unknown (0xffff)
  tx0_serial=
  tx0_subdev_name=FE-TX2
  tx0_subdev_spec=A:A A:B
  tx1_antenna=TX/RX
  tx1_id=Unknown (0xffff)
  tx1_serial=
  tx1_subdev_name=FE-TX1
  tx1_subdev_spec=A:A A:B

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 2 Rx, 2 Tx
  Timestamps: YES
  Clock sources: internal, external, gpsdo
[... 117 lines cut ...]
       [key=WIRE, type=string, options=(sc8, sc16)]
     * Peak value - The peak value for scaling in complex byte mode.
       [key=peak, default=1.0, type=float]
     * Send frame buffer size - The size an individual datagram or frame in bytes.
       [key=send_frame_size, units=bytes, type=int]
     * Send number of buffers - The number of available buffers.
       [key=num_send_frames, units=buffers, type=int]
  Antennas: TX/RX
  Corrections: DC offset, IQ balance
  Full gain range: [0, 89.75, 0.25] dB
    PGA gain range: [0, 89.75, 0.25] dB
  Full freq range: [42, 6008] MHz
    RF freq range: [42, 6008] MHz
    BB freq range: [-8, 8] MHz
  Tune args:
     * LO Offset - Tune the LO with an offset and compensate with the baseband CORDIC.
       [key=OFFSET, units=Hz, default=0.0, type=float, range=[-8e+06, 8e+06, 0.00372529]]
     * BB - Specify a specific value for this component or IGNORE to skip tuning it.
       [key=BB, units=Hz, default=DEFAULT, type=float, range=[-8e+06, 8e+06, 0.00372529], options=(DEFAULT, IGNORE)]
  Sample rates: 0.03125, 0.0314961, 0.031746, 0.032, 0.0322581, ..., 3.2, 4, 5.33333, 8, 16 MSps
  Filter bandwidths: [0.2, 56] MHz
  Sensors: temp, lo_locked
```
Do I need to set anything for the stream settings section?
[image]