pothosware/SoapySDR #167: SoapySDRUtil remembers and uses the cmake install path as its 'root path'

unixpunk, 2018-05-21:
First off, thanks for the software and work on this, very useful and flexible.

In cross-compiling SoapySDR I noticed that since I'm using a full path to a staging directory as my cmake install path, SoapySDRUtil seems to remember this and use it as its root for things, including modules.  Example:
# SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.0-g69c16e98
API Version: v0.7.0
ABI Version: v0.7
Install root: /home/steve/Desktop/SDR/Pluto/staging
Search path: /home/steve/Desktop/SDR/Pluto/staging/lib/SoapySDR/modules0.7
Module found: /home/steve/Desktop/SDR/Pluto/staging/lib/SoapySDR/modules0.7/libPlutSDRSupport.so (dd6686e)
Module found: /home/steve/Desktop/SDR/Pluto/staging/lib/SoapySDR/modules0.7/libremoteSupport.so  (0.5.0-fa71b4e)
Available factories... plutosdr, remote
Available converters...

I'm not sure a better option, but since I need to copy these binaries and have them run on a completely different system, I've had to work around this by creating that staging directory path tree and linking it to /.  Might be related to https://github.com/pothosware/SoapySDR/issues/65 in that if I could set this path on execution or something then I wouldn't need to 'fake' its existence.

Thanks!

guruofquality, 2018-05-21:
When transplanting the installation there is an environment variable: https://github.com/pothosware/SoapySDR/blob/master/CMakeLists.txt#L40 (SOAPY_SDR_ROOT by default) Snap packages use this method.

Sometimes when you know the root and the package will be DIST installed, you can pass `-DSOAPY_SDR_ROOT=` like brew recipe does: https://github.com/pothosware/homebrew-pothos/blob/master/soapysdr.rb#L34

Finally windows can look relative the dll, which is also pretty useful. Ideal I think: https://github.com/pothosware/SoapySDR/blob/master/lib/Modules.in.cpp#L45

unixpunk, 2018-05-21:
Thanks, I think I did use that variable originally (been a while since initially doing the cmake) but I think had to set it to the local path on the cross-compiler because it wasn't finding libs or plugins or something, don't recall exactly now.  I will be running through this entire process again for a tutorial and will test/confirm my results and if I'm still not successful, I can re-open this issue unless you have another/preferred method to discuss.  Feel free to close, no ETA when I'll start from scratch again yet.  Thanks again!

guruofquality, 2018-05-22:
Sure, let me know if something comes up

closed 2018-05-22 by guruofquality