pothosware/SoapySDR #168: Issue building on Centos 7

garengllc, 2018-06-07:
I am pretty sure things used to build on my system, but I just tried a clean install and it didn't work.
I do a git clone, cd SoapySDR, mkdir build, cd build, cmake ../, make VERBOSE=1

It bombs out about 94% of the way through.  Here is what I see:
`[ 91%] Built target docs
make -f python/CMakeFiles/_SoapySDR.dir/build.make python/CMakeFiles/_SoapySDR.dir/depend
make[2]: Entering directory `/tmp/SoapySDR/build'
cd /tmp/SoapySDR/build && /usr/local/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/SoapySDR /tmp/SoapySDR/python /tmp/SoapySDR/build /tmp/SoapySDR/build/python /tmp/SoapySDR/build/python/CMakeFiles/_SoapySDR.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/tmp/SoapySDR/build'
make -f python/CMakeFiles/_SoapySDR.dir/build.make python/CMakeFiles/_SoapySDR.dir/build
make[2]: Entering directory `/tmp/SoapySDR/build'
[ 94%] Building CXX object python/CMakeFiles/_SoapySDR.dir/SoapySDRPYTHON_wrap.cxx.o
cd /tmp/SoapySDR/build/python && /usr/bin/c++   -D_SoapySDR_EXPORTS -I/tmp/SoapySDR/include -I/usr/include/python2.7  -fvisibility-inlines-hidden -O3 -DNDEBUG -fPIC   -Wall -Wextra -fvisibility=hidden -std=gnu++11 -o CMakeFiles/_SoapySDR.dir/SoapySDRPYTHON_wrap.cxx.o -c /tmp/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx
/tmp/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx: In function ‘PyObject* _wrap_SoapySDRKwargsList_erase__SWIG_0(PyObject*, PyObject*)’:
/tmp/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx:14489:36: error: no matching function for call to ‘std::vector<std::map<std::basic_string<char>, std::basic_string<char> > >::erase(SwigValueWrapper<__gnu_cxx::__normal_iterator<std::map<std::basic_string<char>, std::basic_string<char> >*, std::vector<std::map<std::basic_string<char>, std::basic_string<char> > > > >&)’
         result = (arg1)->erase(arg2);
                                    ^
/tmp/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx:14489:36: note: candidates are:
In file included from /usr/local/include/c++/4.9.2/vector:64:0,
                 from /tmp/SoapySDR/include/SoapySDR/Types.hpp:14,
                 from /tmp/SoapySDR/include/SoapySDR/Modules.hpp:15,
                 from /tmp/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx:3096:
/usr/local/include/c++/4.9.2/bits/stl_vector.h:1146:7: note: std::vector<_Tp, _Alloc>::iterator std::vector<_Tp, _Alloc>::erase(std::vector<_Tp, _Alloc>::const_iterator) [with _Tp = std::map<std::basic_string<char>, std::basic_string<char> >; _Alloc = std::allocator<std::map<std::basic_string<char>, std::basic_string<char> > >; std::vector<_Tp, _Alloc>::iterator = __gnu_cxx::__normal_iterator<std::map<std::basic_string<char>, std::basic_string<char> >*, std::vector<std::map<std::basic_string<char>, std::basic_string<char> > > >; typename std::_Vector_base<_Tp, _Alloc>::pointer = std::map<std::basic_string<char>, std::basic_string<char> >*; std::vector<_Tp, _Alloc>::const_iterator = __gnu_cxx::__normal_iterator<const std::map<std::basic_string<char>, std::basic_string<char> >*, std::vector<std::map<std::basic_string<char>, std::basic_string<char> > > >; typename __gnu_cxx::__alloc_traits<typename std::_Vector_base<_Tp, _Alloc>::_Tp_alloc_type>::const_pointer = const std::map<std::basic_string<char>, std::basic_string<char> >*]
       erase(const_iterator __position)
       ^
/usr/local/include/c++/4.9.2/bits/stl_vector.h:1146:7: note:   no known conversion for argument 1 from ‘SwigValueWrapper<__gnu_cxx::__normal_iterator<std::map<std::basic_string<char>, std::basic_string<char> >*, std::vector<std::map<std::basic_string<char>, std::basic_string<char> > > > >’ to ‘std::vector<std::map<std::basic_string<char>, std::basic_string<char> > >::const_iterator {aka __gnu_cxx::__normal_iterator<const std::map<std::basic_string<char>, std::basic_string<char> >*, std::vector<std::map<std::basic_string<char>, std::basic_string<char> > > >}’
/usr/local/include/c++/4.9.2/bits/stl_vector.h:1173:7: note: std::vector<_Tp, _Alloc>::iterator std::vector<_Tp, _Alloc>::erase(std::vector<_Tp, _Alloc>::const_iterator, std::vector<_Tp, _Alloc>::const_iterator) [with _Tp = std::map<std::basic_string<char>, std::basic_string<char> >; _Alloc = std::allocator<std::map<std::basic_string<char>, std::basic_string<char> > >; std::vector<_Tp, _Alloc>::iterator = __gnu_cxx::__normal_iterator<std::map<std::basic_string<char>, std::basic_string<char> >*, std::vector<std::map<std::basic_string<char>, std::basic_string<char> > > >; typename std::_Vector_base<_Tp, _Alloc>::pointer = std::map<std::basic_string<char>, std::basic_string<char> >*; std::vector<_Tp, _Alloc>::const_iterator = __gnu_cxx::__normal_iterator<const std::map<std::basic_string<char>, std::basic_string<char> >*, std::vector<std::map<std::basic_string<char>, std::basic_string<char> > > >; typename __gnu_cxx::__alloc_traits<typename std::_Vector_base<_Tp, _Alloc>::_Tp_alloc_type>::const_pointer = const std::map<std::basic_string<char>, std::basic_string<char> >*]
       erase(const_iterator __first, const_iterator __last)
       ^
/usr/local/include/c++/4.9.2/bits/stl_vector.h:1173:7: note:   candidate expects 2 arguments, 1 provided
/tmp/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx: In function ‘PyObject* _wrap_SoapySDRKwargsList_erase__SWIG_1(PyObject*, PyObject*)’:
/tmp/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx:14558:41: error: no matching function for call to ‘std::vector<std::map<std::basic_string<char>, std::basic_string<char> > >::erase(SwigValueWrapper<__gnu_cxx::__normal_iterator<std::map<std::basic_string<char>, std::basic_string<char> >*, std::vector<std::map<std::basic_string<char>, std::basic_string<char> > > > >&, SwigValueWrapper<__gnu_cxx::__normal_iterator<std::map<std::basic_string<char>, std::basic_string<char> >*, std::vector<std::map<std::basic_string<char>, std::basic_string<char> > > > >&)’
         result = (arg1)->erase(arg2,arg3);
                                         ^
/tmp/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx:14558:41: note: candidates are:
In file included from /usr/local/include/c++/4.9.2/vector:64:0,
                 from /tmp/SoapySDR/include/SoapySDR/Types.hpp:14,
                 from /tmp/SoapySDR/include/SoapySDR/Modules.hpp:15,
                 from /tmp/SoapySDR/build/python/SoapySDRPYTHON_wrap.cxx:3096:
/usr/local/include/c++/4.9.2/bits/stl_vector.h:1146:7: note: std::vector<_Tp, _Alloc>::iterator std::vector<_Tp, _Alloc>::erase(std::vector<_Tp, _Alloc>::const_iterator) [with _Tp = std::map<std::basic_string<char>, std::basic_string<char> >; _Alloc = std::allocator<std::map<std::basic_string<char>, std::basic_string<char> > >; std::vector<_Tp, _Alloc>::iterator = __gnu_cxx::__normal_iterator<std::map<std::basic_string<char>, std::basic_string<char> >*, std::vector<std::map<std::basic_string<char>, std::basic_string<char> > > >; typename std::_Vector_base<_Tp, _Alloc>::pointer = std::map<std::basic_string<char>, std::basic_string<char> >*; std::vector<_Tp, _Alloc>::const_iterator = __gnu_cxx::__normal_iterator<const std::map<std::basic_string<char>, std::basic_string<char> >*, std::vector<std::map<std::basic_string<char>, std::basic_string<char> > > >; typename __gnu_cxx::__alloc_traits<typename std::_Vector_base<_Tp, _Alloc>::_Tp_alloc_type>::const_pointer = const std::map<std::basic_string<char>, std::basic_string<char> >*]
       erase(const_iterator __position)
       ^
/usr/local/include/c++/4.9.2/bits/stl_vector.h:1146:7: note:   candidate expects 1 argument, 2 provided
/usr/local/include/c++/4.9.2/bits/stl_vector.h:1173:7: note: std::vector<_Tp, _Alloc>::iterator std::vector<_Tp, _Alloc>::erase(std::vector<_Tp, _Alloc>::const_iterator, std::vector<_Tp, _Alloc>::const_iterator) [with _Tp = std::map<std::basic_string<char>, std::basic_string<char> >; _Alloc = std::allocator<std::map<std::basic_string<char>, std::basic_string<char> > >; std::vector<_Tp, _Alloc>::iterator = __gnu_cxx::__normal_iterator<std::map<std::basic_string<char>, std::basic_string<char> >*, std::vector<std::map<std::basic_string<char>, std::basic_string<char> > > >; typename std::_Vector_base<_Tp, _Alloc>::pointer = std::map<std::basic_string<char>, std::basic_string<char> >*; std::vector<_Tp, _Alloc>::const_iterator = __gnu_cxx::__normal_iterator<const std::map<std::basic_string<char>, std::basic_string<char> >*, std::vector<std::map<std::basic_string<char>, std::basic_string<char> > > >; typename __gnu_cxx::__alloc_traits<typename std::_Vector_base<_Tp, _Alloc>::_Tp_alloc_type>::const_pointer = const std::map<std::basic_s [... cut ...]
         (arg1)->insert(arg2,arg3,(std::vector< std::map< std::string,std::string,std::less< std::string >,std::allocator< std::pair< std::string const,std::string > > > >::value_type const &)*arg4);
                                                                                                                                                                                                     ^
make[2]: *** [python/CMakeFiles/_SoapySDR.dir/SoapySDRPYTHON_wrap.cxx.o] Error 1
make[2]: Leaving directory `/tmp/SoapySDR/build'
make[1]: *** [python/CMakeFiles/_SoapySDR.dir/all] Error 2
make[1]: Leaving directory `/tmp/SoapySDR/build'
make: *** [all] Error 2
`

guruofquality, 2018-06-07:
Looks like swig is generating wrappers that are not compatible with the stl supported by the compiler. Like it cant find std::map erase/insert variants. So you might just want a more recent compiler or another way to get around this is to disable the python bindings (if you arent using them)

garengllc, 2018-06-08:
For reasons unknown to me, this seems to build fine on a fresh Centos7 box, so there must be something wrong with my current build system.  I'll close this issue out as it was my fault, not Soapy's. Thanks.

closed 2018-06-08 by garengllc