pothosware/SoapySDR #169: Please add cmake options allowing to build C++ and python libraries separately

yurivict, 2018-06-19:
I need to build C++ once, and python version several times for several python versions.
This is for the FreeBSD ports top be able to run with any python versions.

guruofquality, 2018-06-20:
Yea, I agree that python bindings have been somewhat annoying for reasons like this. With the pothos library, the python bindings are already a separate project, and do get built multiple times in some of the different build environments.

I was thinking that I could make the python/CMakeLists.txt its own top level project, so you could build only python bindings (for whatever flags were passed in) and it would use the existing soapysdr development installation.

So the proposal:
```
cmake src/SoapySDR/ -DENABLE_PYTHON=OFF -DENABLE_PYTHON3=OFF #for building main library

cmake src/SoapySDR/python -DPYTHON_<ETC, ETC> #for python bindings
```

Just to double check, you are building for more than python2 and 3. Like you need bindings for 2.7, 3.5, 3.6, etc. Because the existing build can do both 2 and 3, typically this is for a debian-like setup with an official python2 and python3 version (for whatever 2.x or 3.y they may be).

yurivict, 2018-06-20:
Yes, the port can be build for py27, py34, py35, py36, etc.

I would suggest to have the ```-DPYTHON_CMD=/path/to/python``` variable that would come with ```-DENABLE_PYTHON=ON```, and to read the version and everything else from there.

guruofquality, 2018-06-20:
If you pass -DPYTHON_EXECUTABLE=/path/to/python and `-DUSE_PYTHON_CONFIG=ON`, the cmake actually will use /path/to/python-config executable to query the compiler libs and includes. I added that because cmake's default findpythonlibs is pretty inconsistent, but asking python itself where to find things tends to do better.

In short, all you need is the python executable and you should be able to have the build figure out the rest in terms of flags and libs.

Note to self: make USE_PYTHON_CONFIG default when the config executable is present.

guruofquality, 2018-06-20:
A little awkward, but it looks like this use case already exists. Example:

```
cmake ~/src/SoapySDR/python -DENABLE_LIBRARY=ON -DUSE_PYTHON_CONFIG=ON -DPYTHON_EXECUTABLE=/usr/bin/python3
make
sudo make install
```

yurivict, 2018-06-20:
But doesn't it still rebuild all C++ code?

guruofquality, 2018-06-20:
nope, it just uses the installed library found through `find_package(SoapySDR NO_MODULE REQUIRED)`

guruofquality, 2018-06-24:
> You should create a totally separate project for the python binding. It should be python distutils-based, with setup.py. This way python modules work best. cmake isn't great for python.

Any volunteers? Maybe distutils is better, but the current build works, python bindings can be build separately as requested, and you can specify any version of python (use -DPYTHON_EXECUTABLE) and installation directory (use -DPYTHON_INSTALL_DIR), and it models similar projects in the community like gnuradio, gr-osmosdr.

To build without cmake, just replicate the swig generation, compilation, and installation. Also mind the flag for size_t vs unsigned int. But is totally possible.

## Building Python libs separately on maint branch or release tags

Anyway wrapping up this issue, it tuned out that you can already do this with the current build, so for a release tag or maint branch, use: 

```
#build the main library without python bindings
mkdir build-main-libs && cd build-main-libs
cmake ~/src/SoapySDR -DENABLE_PYTHON=OFF -DENABLE_PYTHON3=OFF
make && sudo make install

#build python bindings for 2.7
mkdir build-python2.7 && cd build-python2.7
cmake ~/src/SoapySDR/python -DENABLE_LIBRARY=ON -DUSE_PYTHON_CONFIG=ON -DPYTHON_EXECUTABLE=/usr/bin/python2.7
make && sudo make install

#build python bindings for 3.5
mkdir build-python3.5 && cd build-python3.5
cmake ~/src/SoapySDR/python -DENABLE_LIBRARY=ON -DUSE_PYTHON_CONFIG=ON -DPYTHON_EXECUTABLE=/usr/bin/python3.5
make && sudo make install

#build python bindings for 3.6
mkdir build-python3.6 && cd build-python3.6
cmake ~/src/SoapySDR/python -DENABLE_LIBRARY=ON -DUSE_PYTHON_CONFIG=ON -DPYTHON_EXECUTABLE=/usr/bin/python3.6
make && sudo make install
```

## Building Python libs separately on master branch

I wanted to make this simpler, so I added two changes to master branch: https://github.com/pothosware/SoapySDR/commit/66306d24e07d2cd67f292286f3f460734700c234

USE_PYTHON_CONFIG is on by default and ENABLE_LIBRARY is not needed when using the python/ subdirectory as a top level project. Basically, the flags are just easier, and its backwards compatible, so specifying the flags mentioned previously wont hurt anything.

```
#build the main library without python bindings
mkdir build-main-libs && cd build-main-libs
cmake ~/src/SoapySDR -DENABLE_PYTHON=OFF -DENABLE_PYTHON3=OFF
make && sudo make install

#build python bindings for 2.7
mkdir build-python2.7 && cd build-python2.7
cmake ~/src/SoapySDR/python -DPYTHON_EXECUTABLE=/usr/bin/python2.7
make && sudo make install

#build python bindings for 3.5
mkdir build-python3.5 && cd build-python3.5
cmake ~/src/SoapySDR/python -DPYTHON_EXECUTABLE=/usr/bin/python3.5
make && sudo make install

#build python bindings for 3.6
mkdir build-python3.6 && cd build-python3.6
cmake ~/src/SoapySDR/python -DPYTHON_EXECUTABLE=/usr/bin/python3.6
make && sudo make install
```

I'm closing the issue because the original request has been addressed, you can definitely `build the C++ library once, and python version several times for several python versions`.

Configuring a project with flags, make, and make install should be pretty standard, so hopefully this fits nicely with ports. Yes, its not distutils, but I would like to know if there is actually a build problem with the methodology suggested above for freebsd ports. Or if the cmake command or make command fails with the default unpatched master branch, maint branch, or release tags. Thanks!

closed 2018-06-24 by guruofquality

yurivict, 2018-06-24:
It always rebuilds the C library with the python binding. I followed your instructions, and it still does.

guruofquality, 2018-06-24:
> It always rebuilds the C library with the python binding. I followed your instructions, and it still does.

Can you share all commands run and the verbose/output of cmake and make commands?

yurivict, 2018-06-24:
Here is the build log in the python binding port:
[log.txt](https://github.com/pothosware/SoapySDR/files/2131057/log.txt)

This line appears to unconditionally install headers:
```
install(DIRECTORY include/SoapySDR DESTINATION include)
```
This line installs cmake files unconditionally
```
install(FILES ${cmake_files} DESTINATION share/cmake/${PROJECT_NAME})
```
Library is also always built and installed.

guruofquality, 2018-06-24:
I cant actually see the cmake command in the log, but you are passing the wrong directory. For the python only builds not `cmake `~/src/SoapySDR`, its `cmake ~/src/SoapySDR/python`