pothosware/SoapySDR #170: Build breaks on 32-bit systems: error: redefinition of 'traits<unsigned int>'

yurivict, 2018-06-21:
The FreeBSD port breaks on i386 systems:
```
python3/SoapySDRPYTHON_wrap.cxx:6096:22: error: redefinition of 'traits<unsigned int>'
  template <> struct traits< size_t > {
                     ^~~~~~~~~~~~~~~~
python3/SoapySDRPYTHON_wrap.cxx:4961:22: note: previous definition is here
  template <> struct traits< unsigned int > {
                     ^
python3/SoapySDRPYTHON_wrap.cxx:6100:23: error: redefinition of 'traits_asval<unsigned int>'
  template <>  struct traits_asval< size_t > {
                      ^~~~~~~~~~~~~~~~~~~~~~
python3/SoapySDRPYTHON_wrap.cxx:4965:23: note: previous definition is here
  template <>  struct traits_asval< unsigned int > {
                      ^
python3/SoapySDRPYTHON_wrap.cxx:6106:23: error: redefinition of 'traits_from<unsigned int>'
  template <>  struct traits_from< size_t > {
                      ^~~~~~~~~~~~~~~~~~~~~
python3/SoapySDRPYTHON_wrap.cxx:4971:23: note: previous definition is here
  template <>  struct traits_from< unsigned int > {
                      ^
python3/SoapySDRPYTHON_wrap.cxx:6116:22: error: redefinition of 'traits<std::__1::vector<unsigned int, std::__1::allocator<unsigned int> > >'
        template <>  struct traits<std::vector< size_t, std::allocator< size_t > > > {
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
python3/SoapySDRPYTHON_wrap.cxx:5085:22: note: previous definition is here
        template <>  struct traits<std::vector< unsigned int, std::allocator< unsigned int > > > {
                            ^
```

guruofquality, 2018-06-21:
Because size_t varies in type, sometimes theres multiple definitions generated for vectors of the same integer types, etc. Freebsd may have a different typedef for it that not considered. Can you tell me how size_t is defined on freebsd i386?  It may not need this ifdef stuff below, or maybe its different.

A similar work around I already use because of this:
* https://github.com/pothosware/SoapySDR/blob/master/python/SoapySDR.i#L48
* https://github.com/pothosware/SoapySDR/blob/master/python/CMakeLists.txt#L105

yurivict, 2018-06-21:
```typedef uint32_t  size_t```

guruofquality, 2018-06-21:
Whats the typedef for the uint32_t? Like unsigned int, unsigned, unsigned long, etc? Looking for the native primitive type. It seems like its unsigned int since the conflict is between size_t and unsigned vectors, but the SIZE_T_IS_UNSIGNED_INT isnt being set.

Can you also try forcing the line in the cmakelists.txt `list(APPEND CMAKE_SWIG_FLAGS -DSIZE_T_IS_UNSIGNED_INT)` to always run, without the 'if' clauses.

Also whats the cmake output, does it show the test for SIZE_T_IS_UNSIGNED_INT as passing or failing? I'm worried its supposed to be passing and its not.

If it fails, can you attach `./CMakeFiles/CMakeError.log`

yurivict, 2018-06-21:
```typedef unsigned int            __uint32_t;```

I will answer other questions when I get the failure log gain. I couldn't reproduce it myself.

yurivict, 2018-06-21:
I recommend you only define template instantiations for explicitly different types - ex. by size. Because size_t is always one of them.

guruofquality, 2018-06-21:
Well its swig's template generation, so the only thing I can tell it to do is to not generate certain exports when they overlap with size_t.

I think the jist is that this C code is failing to compile:
```
    #include <cstddef>
    int main() {
    size_t *x = (unsigned int *)(NULL);
    return 0; }"
```

It should compile since size_t is unsigned int. So I will have to hear back about the log

yurivict, 2018-06-23:
The failure log: http://beefy10.nyi.freebsd.org/data/111i386-default/473110/logs/py36-SoapySDR-0.6.1.71.log

guruofquality, 2018-06-23:
Well its not the cmake error log, but it turned up something. Its building the `Python3, python3 bindings`  and not the regular python ones. The trouble is that the regular one on the python folder does the check for this unsigned size_t thing. I don't see the regular `Python, python bindings` even in there disabled or enabled, is this build patched?

```
- 
-- ######################################################
-- ## SoapySDR enabled features
-- ######################################################
-- 
 * Library, runtime library v0.7.0-unknown
 * Tests, library unit tests
 * Python3, python3 bindings

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
-- 
 * Docs, doxygen documentation
```

Anyway, I think thats the cause of your problem. If you are doing the split build like you mentioned in the other issue #169.

* build the library with `cmake ~/src/SoapySDR -DENABLE_PYTHON=OFF -DENABLE_PYTHON3=OFF`
* build the python bindings with `cmake ~/src/SoapySDR/python -DENABLE_LIBRARY=ON -DUSE_PYTHON_CONFIG=ON -DPYTHON_EXECUTABLE=/usr/bin/python3.4.5`

guruofquality, 2018-06-23:
Any version 2.7 and up should be fine. I think patching is what got you into this mess.

* What happened to https://github.com/pothosware/SoapySDR/issues/169?
* Are you trying to have one library build and several separate builds for each python version?,
*  or are you trying to build all python versions with the library build?

I get that the default python cmake isnt the smartest, but it can be overridden with flags for any version, and that the build only supports two version of python simultaneously... but I think your strategy from #169 was pretty good, and you dont need any patches for that to work.

I'm closing this issue, because as far as I can tell the error is because you removed an important check in the build that defines `SIZE_T_IS_UNSIGNED_INT`: https://github.com/pothosware/SoapySDR/blob/master/python/CMakeLists.txt#L100

closed 2018-06-23 by guruofquality

yurivict, 2018-06-23:
I don't think I disabled that line with ```SIZE_T_IS_UNSIGNED_INT```. But anyway, I will manually re-enable this define and see if this will fix it.

yurivict, 2018-06-23:
> build the python bindings with cmake ~/src/SoapySDR/python -DENABLE_LIBRARY=ON -DUSE_PYTHON_CONFIG=ON -DPYTHON_EXECUTABLE=/usr/bin/python3.4.5

This way it still always builds C library in a python module. Without patches, it doesn't even know about python 3.6.

So I will just disable i386 for now, until #169 is fixed.

guruofquality, 2018-06-24:
> I don't think I disabled that line with SIZE_T_IS_UNSIGNED_INT. But anyway, I will manually re-enable this define and see if this will fix it.

Well the output from the python/ subdirectory was missing, and critically this was missing from the output of cmake, (it should say pass or fail):

```
-- Performing Test SIZE_T_IS_UNSIGNED_INT
-- Performing Test SIZE_T_IS_UNSIGNED_INT - Failed
```

> No, just defining it breaks the build.

What was the diff for this change, and what was the build output?

Did you use the following line as in `python/CMakeLists.txt`?

`list(APPEND CMAKE_SWIG_FLAGS -DSIZE_T_IS_UNSIGNED_INT)`?

> This way it still always builds C library in a python module.

What do you mean?

> Without patches, it doesn't even know about python 3.6.

What's the output when you tell it to use python3.6? When you substitute in the real path to the 3.6 python executable?

`cmake ~/src/SoapySDR/python -DENABLE_LIBRARY=ON -DUSE_PYTHON_CONFIG=ON -DPYTHON_EXECUTABLE=/usr/bin/python3.6`

yurivict, 2018-06-24:
Using ```python/python3``` subdirectory solves the problem of C library always being built. Thank you!

------

Now there's only a problem of i386 build left. When on i386 systems I add the argument to cmake ```-DSIZE_T_IS_UNSIGNED_INT:BOOL=ON```, builds still fails for python-3.6 on i386.

You need to calculate ```SIZE_T_IS_UNSIGNED_INT``` automatically, because there are more architectures than amd64/i386. It will most certainly fail there.

guruofquality, 2018-06-24:
> Using python/python3 subdirectory solves the problem of C library always being built. Thank you!

Don't use the python3 directory, its missing important stuff and its there for the joint python2+3 debian builds and similar OS.

Just like I said in #169 always `cmake SoapySDR/python -DPYTHON_EXECUTABLE=/usr/bin/python3.5`

> -DSIZE_T_IS_UNSIGNED_INT:BOOL=ON

This does not set compiler flags, so it basically does nothing

yurivict, 2018-06-25:
> Just like I said in #169 always cmake SoapySDR/python -DPYTHON_EXECUTABLE=/usr/bin/python3.5

This works, thanks!

I still have this patch:
```
--- ../python/CMakeLists.txt.orig       2018-06-19 07:11:01 UTC
+++ ../python/CMakeLists.txt
@@ -121,7 +121,7 @@ set(CMAKE_SWIG_FLAGS ${CMAKE_SWIG_FLAGS}
 ########################################################################
 include(FeatureSummary)
 include(CMakeDependentOption)
-cmake_dependent_option(ENABLE_PYTHON "Enable python bindings" ON "ENABLE_LIBRARY;SWIG_FOUND;PYTHONINTERP_FOUND;PYTHONLIBS_FOUND;PYTHON_DEBUG_OK;PYTHON_VERSION_MATCH" OFF)
+option(ENABLE_PYTHON "Enable python bindings" ON)
 add_feature_info(Python ENABLE_PYTHON "python bindings v${PYTHON_VERSION_STRING}")
 if (NOT ENABLE_PYTHON)
     return()
```
Otherwise ```ENABLE_PYTHON``` doesn't get enabled because it has preconditions that aren't met.

----

Plus you need to fix this i386 failure.

This C++ check can determine type equivalence:
```
template<typename T>
struct S {
 enum {TypeIsUInt=false};
};

template<>
struct S<unsigned int> {
 enum {TypeIsUInt=true};
};

#include <stdio.h>

int main() {
  printf("SIZE_T_IS_UNSIGNED_INT=%s\n", S<size_t>::TypeIsUInt ? "YES" : "NO");
  return 0;
}
```

You should plug this test into a cmake check to compute ```SIZE_T_IS_UNSIGNED_INT```.

guruofquality, 2018-06-25:
> Otherwise ENABLE_PYTHON doesn't get enabled because it has preconditions that aren't met.

Which one of these variables is not set?

ENABLE_LIBRARY;SWIG_FOUND;PYTHONINTERP_FOUND;PYTHONLIBS_FOUND;PYTHON_DEBUG_OK;PYTHON_VERSION_MATCH

> Plus you need to fix this i386 failure.

Are you saying the current check is not functioning? This is why I was asking about the cmake error log `./CMakeFiles/CMakeError.log`

If you are not getting that SWIG flag set, then the check is erroring out and I need to see that log output to fix the problem. Remember, the check should pass and gcc should build successfully on i386 systems.

For example, on x64 processors, the check should fail, you will see something like this:
```
Building CXX object CMakeFiles/cmTC_7340e.dir/src.cxx.o
/usr/local/bin/c++     -fvisibility-inlines-hidden -DSIZE_T_IS_UNSIGNED_INT   -o CMakeFiles/cmTC_7340e.dir/src.cxx.o -c /home/jblum/user/SoapySDR/CMakeFiles/CMakeTmp/src.cxx
/home/user/build/SoapySDR/CMakeFiles/CMakeTmp/src.cxx: In function ‘int main()’:
/home/user/build/SoapySDR/CMakeFiles/CMakeTmp/src.cxx:4:37: error: cannot convert ‘unsigned int*’ to ‘size_t* {aka long unsigned int*}’ in initialization
     size_t *x = (unsigned int *)(NULL);
```

> This C++ check can determine type equivalence:

The current build is using a compiler check, so its not running the executable, its just checking if gcc completes. The snippet you sent should always compile reguardless of size.

I don't know if this was just for fun, I agree that there there is probably a more c++ friendly way where the check can fail with multiple redundant template instantiations just like it does in the swig generated code. 

Lets see the CMakeError.log first before overcomplicating it.

yurivict, 2018-06-25:
I take it back, now it builds fine on i386. Patches might have caused it.

Thanks!