pothosware/SoapySDR #171: Possible Exe/Dll Mismatch using Soapy (Windows 10/LimeSDR)

Ken98045, 2018-06-24:
Using SoapySDRUtil --find I get the following: "Error: EnvironmentError: OSError: LoadLibrary failed to load "C:\Program Files\PothosSDR\lib\uhd\modules\umtrx.dll"
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106300; UHD_3.11.0.1-0-ge00f1f37"

This looks like a version .dll/.exe mismatch error, but I can't tell if it is on your side or mine.  I have tried uninstalling and reinstalling both the Lime driver and the entire Pothos environment.  

Any help would be appreciated.

guruofquality, 2018-06-24:
its not really a critical error. But you should uninstall before you re-install, otherwise old modules are laying around to conflict. umtrx isnt in the build anymore since its not maintained: https://github.com/fairwaves/UHD-Fairwaves/issues/21

closed 2018-06-24 by guruofquality

Ken98045, 2018-06-25:
I did use the uninstall shortcut, but I was too lazy (well, it was 3:am)  to notice that the uninstall doesn't (or didn't for me) actually delete existing files under Program FIles.  Once I nuked the folder and re-installed things work OK.  However, for whatever reason GQRX won't work.  GQRX was getting an error related to that DLL, but that problem at least went away with my cleaner install.  That's probably not your problem, but other SDR apps do work, so it might be nice if someone could do a clean install and double check that.  Windows GQRX seems to be the unwanted step child, but....

guruofquality, 2018-06-25:
Sometimes the uninstall cant remove all of the files because something running/crashed in the background.

The umtrx load error -- Its more of a warning print that it couldnt load a hardware support module, I dont see it being the breaking issue. Maybe though...

GQRX itself is pretty decent code and easy to build, but it has the disadvantage of having gnuradio and grosmosdr as somewhat bulky dependencies. I will check it next time I do the build