pothosware/SoapySDR #172: Cross compiling SoapySDR and modules

lpzantony, 2018-06-25:
Hello,

My name is Antony and I am trying to develop my own headless spectrum analyser software from HackRF and/or RTL-SDR USB sticks. It's basically the only SDR hardware I currently have. Spectrum data is retrieved using TCP. I decided to use SoapySDR to benefit from the hardware abstraction it offers. 

I am making great progress on my desktop PC version, but now I would like to run it on a raspberry-like ARM based board running Debian Jessy. I would like to keep compiling my software on my computer, so I figured out I had to cross-compile my dependencies which are SoapySDR and the modules I use : SoapyHackRF and SoapyRTLSDR. SoapySDR cross-compiles easily but not the modules.
I used this command to cross compile SoapySDR inside the cloned repo in a `build` directory:
```bash
cmake \
    -D CMAKE_C_COMPILER=/usr/bin/arm-linux-gnueabihf-gcc \
    -D CMAKE_CXX_COMPILER=/usr/bin/arm-linux-gnueabihf-g++ \
    -D CMAKE_AR=/usr/bin/arm-linux-gnueabihf-ar \
    -D CMAKE_C_COMPILER_AR=/usr/bin/arm-linux-gnueabihf-gcc-ar \
    -D CMAKE_C_COMPILER_RANLIB=/usr/bin/arm-linux-gnueabihf-gcc-ranlib \
    -D CMAKE_LINKER=/usr/bin/arm-linux-gnueabihf-ld \
    -D CMAKE_NM=/usr/bin/arm-linux-gnueabihf-nm \
    -D CMAKE_OBJCOPY=/usr/bin/arm-linux-gnueabihf-objcopy \
    -D CMAKE_OBJDUMP=/usr/bin/arm-linux-gnueabihf-objdump \
    -D CMAKE_RANLIB=/usr/bin/arm-linux-gnueabihf-ranlib \
    -D CMAKE_STRIP=/usr/bin/arm-linux-gnueabihf-strip \
    ..
```
I get my .so file for ARM successfully. 

For SoapyHackRF using the same command for cmake fails at generating Makefile because cmake can't find the hackRF library nor its `include` folder :
```bash
-- The CXX compiler identification is GNU 8.1.0
-- Check for working CXX compiler: /usr/bin/arm-linux-gnueabihf-g++
-- Check for working CXX compiler: /usr/bin/arm-linux-gnueabihf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.4.2") 
-- Could NOT find LIBHACKRF (missing: LIBHACKRF_LIBRARY LIBHACKRF_INCLUDE_DIR) 
CMake Error at CMakeLists.txt:17 (message):
  HackRF development files not found...
```

The result is the same for SoapyRTLSDR which can't find the  osmocom's `rtl-sdr` library.  
It looks like I'll have to cross-compile every single dependency of those Soapy modules. I wonder if there is an easier way to cross compile my software for ARM?

Oh and my PC is running Manjaro :)

Thank you in advance for your help !

Sincerely

Antony L

unixpunk, 2018-06-25:
Hi Antony, I've some experience here but not with SoapyHackRF- I don't know for sure if the module needs the compiled library and/or just the include dir, but you can test by adding the -DLIBHACKRF_INCLUDE_DIR=<path to libhackrf source include dir>.  If the module does indeed need the built library also then you'll need to first cross-compile libhackrf and 'install' it to a staging directory which you'll then point the option -DLIBHACKRF_LIBRARY=stagingDir/usr/lib (or wherever)

You MIGHT be able to fake this by copying the includes and library from a Pi to your PC and then pointing those options accordingly.

Hope that helps!

unixpunk, 2018-06-25:
Also, you might consider posting here as that is a more appropriate venue, I think, https://groups.google.com/d/forum/pothos-users

lpzantony, 2018-06-26:
Dully, noted, I answered to your advices [here on the Google group](https://groups.google.com/forum/#!topic/pothos-users/VoownBd18FI)

About copying includes and libs from the Pi, that is what I tried in the first place but I got `undefined reference` errors. It was two weeks ago, I'll retry it today to make sure and will post the result on the Google Group.

closed 2018-06-26 by lpzantony