pothosware/SoapySDR #176: ERROR: Unable to find host: No such host is known.

asymcon, 2018-07-13:
Hi,
getting lot of these errors when trying to initialize my LimeSDR mini in Pothos Flow:

[05:57:15.094246] PothosFlow.MainWindow: Welcome to Pothos v0.6.1-PothosSDR-2018.07.04-vc14-x64
[05:57:15.394247] PothosFlow.MainWindow: Initialization complete
[05:57:15.439247] PothosFlow.BlockCache: Failed to query JSON Docs from tcp://127.0.0.1 - Remote Client Error: Pothos::RemoteClient(tcp://127.0.0.1): Timeout: connect timed out: 127.0.0.1:16415
[05:57:45.764000] Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(E:\Programs\PothosSDR\lib\Pothos\modules0.6\proxy\environment\PythonSupport.dll): failed safe load
[05:57:45.769000] Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(E:\Programs\PothosSDR\lib\Pothos\modules0.6\proxy\environment\PythonSupport3.dll): failed safe load
[05:57:53.966000] ::1: ERROR: Unable to find host: No such host is known. 
[05:57:54.186000] : [INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.11.1.0-0-gad6b0935
[05:57:54.886416] PothosFlow.EvalEngine ⇒ 
Detected evaluation thread lock-up. The evaluator will not function:
EvalEngineImpl.cpp:182 void __cdecl EvalEngineImpl::evaluate(void)
  BlockEval.cpp:129 void __cdecl BlockEval::update(void) [SoapySDRSource0]
    BlockEval.cpp:165 bool __cdecl BlockEval::evaluationProcedure(void)
      BlockEval.cpp:265 get overlay

[05:57:56.746419] PothosFlow.EvalEngine: Evaluation thread has recovered. Perhaps a call is taking too long.
[05:58:07.390000] ::1: ERROR: Unable to find host: No such host is known. 
[05:58:20.952000] ::1: ERROR: Unable to find host: No such host is known. 
[05:58:34.481000] ::1: ERROR: Unable to find host: No such host is known. 
[05:58:47.990000] ::1: ERROR: Unable to find host: No such host is known. 
[05:59:01.514000] ::1: ERROR: Unable to find host: No such host is known. 
[05:59:14.918000] ::1: ERROR: Unable to find host: No such host is known. 
[05:59:28.423000] ::1: ERROR: Unable to find host: No such host is known. 
[05:59:41.955000] ::1: ERROR: Unable to find host: No such host is known. 
[05:59:55.782000] ::1: ERROR: Unable to find host: No such host is known. 
[06:00:09.310000] ::1: ERROR: Unable to find host: No such host is known. 

Any idea for a fix?

guruofquality, 2018-07-13:
>  ERROR: Unable to find host: No such host is known.
> Timeout: connect timed out: 127.0.0.1:16415

Ive seen this once before I think, a VPN for some reason setops the ability for the application to resolve 127.0.0.1. Its just trying to talk to the scratch process it just spawned over localhost. Does anything like that sound familar? VPN issue, just a guess..

Do the command line utilities see the LimeSDR?

`LimeUtil --find`
`SoapySDRUtil.exe --find`
`PothosUtil --device-info`

> \environment\PythonSupport3.dll): failed safe load

Its harmless but you dont need to check the boxes unless you are developing python blocks and you shouldn't check both python2 and 3. Its the C++ bindings for python, the usual python bindings are installed regardless.

asymcon, 2018-07-13:
> Do the command line utilities see the LimeSDR?
Yes

> LimeUtil --find
  * [LimeSDR Mini, media=USB 2, module=FT601, serial=1D3AD617CBC3XX, index=0] 
(removed 2 digits from S/N for privacy purposes)

> SoapySDRUtil.exe --find

######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

ERROR: Unable to find host: No such host is known.
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.0; Boost_106700; UHD_3.11.1.
0-0-gad6b0935
Found device 0
  default_input = True
  default_output = True
  device_id = 0
  driver = audio
  label = Default Device

Found device 1
  default_input = False
  default_output = False
  device_id = 2
  driver = audio
  label = Ext Mic (High Definition Audio Device)

Found device 2
  driver = lime
  index = 0
  label = LimeSDR Mini [USB 2] 1D3AD617CBC3B3
  media = USB 2
  module = FT601
  name = LimeSDR Mini
  serial = 1D3AD617CBC3XX

> PothosUtil --device-info
2018-07-13 16:44:35 Pothos.PluginLoader.load: Plugin Module Error: Pothos::Plugi
nModule(E:\Programs\PothosSDR\lib\Pothos\modules0.6\proxy\environment\PythonSupp
ort.dll): failed safe load
2018-07-13 16:44:35 Pothos.PluginLoader.load: Plugin Module Error: Pothos::Plugi
nModule(E:\Programs\PothosSDR\lib\Pothos\modules0.6\proxy\environment\PythonSupp
ort3.dll): failed safe load

>>> Specify --device-info=deviceType for more information...
>>> Available device types in the plugin tree are:
  * windows_logical_processor
  * audio
  * iio
  * soapy

> Its harmless but you dont need to check the boxes unless you are developing python blocks and you shouldn't check both python2 and 3. Its the C++ bindings for python, the usual python bindings are installed regardless.

I installed just python2 for the current install, but the extra dll might have slipped there in the previous one and uninstaller haven't removed it. 

Otherwise, the usual:
OS: Win 7 x64
Disabled windows firewall
No VPNs installed, no VPN services running

asymcon, 2018-07-14:
This log is with removed uhdsupport.dll from "\PothosSDR\lib\**SoapySDR**\modules0.6"
There was no uhdsupport.dll in "pothosSDR\lib\**Pothos**\module0.6\"

[08:10:08.182687] PothosFlow.MainWindow: Welcome to Pothos v0.6.1-PothosSDR-2018.07.04-vc14-x64
[08:10:08.417688] PothosFlow.MainWindow: Initialization complete
[08:10:12.667000] Pothos.PluginLoader.load: Plugin Module Error: Pothos::PluginModule(E:\Programs\PothosSDR\lib\Pothos\modules0.6\proxy\environment\PythonSupport.dll): failed safe load
[08:10:20.704000] ::1: ERROR: Unable to find host: No such host is known. 
[08:10:31.361000] ::1: ERROR: Unable to find host: No such host is known. 
[08:10:32.441000] SoapySDR: Make connection: 'LimeSDR Mini [USB 2] 1D3AD617CBC3B3'
[08:10:33.091000] SoapySDR: Reference clock 40.00 MHz
[08:10:33.096000] SoapySDR: Device name: LimeSDR-Mini
[08:10:33.096000] SoapySDR: Reference: 40 MHz
[08:10:34.016000] SoapySDR: LMS7002M calibration values caching Disable
[08:10:34.681000] SoapySDR: Cannot achieve desired sample rate: rate too low
[08:10:42.174000] ::1: ERROR: Unable to find host: No such host is known. 
[08:10:52.897000] ::1: ERROR: Unable to find host: No such host is known. 
[08:10:53.877778] default: libpng warning: iCCP: known incorrect sRGB profile
[08:11:02.017803] default: libpng warning: iCCP: known incorrect sRGB profile
[08:11:03.702000] ::1: ERROR: Unable to find host: No such host is known. 
[08:11:12.584000] SoapySDR: Rx calibration: MCU error 5 (Loopback signal weak: not connected/insufficient gain?)
[08:11:13.309822] default: libpng warning: iCCP: known incorrect sRGB profile
[08:11:14.424000] ::1: ERROR: Unable to find host: No such host is known. 
[08:11:25.527000] ::1: ERROR: Unable to find host: No such host is known. 
[08:11:30.167857] default: libpng warning: iCCP: known incorrect sRGB profile
[08:11:36.346000] ::1: ERROR: Unable to find host: No such host is known. 
[08:11:47.285000] ::1: ERROR: Unable to find host: No such host is known. 

Even when I create this simple diagram:
[image]
Spectrum isn't shown upon pressing F6

guruofquality, 2018-07-14:
> Originally, the issue was even when I created this simple diagram:
> Spectrum isn't shown upon pressing F6

Did you insert the graphical widget?

https://github.com/pothosware/PothosFlow/wiki/Tutorial#insert-graphical-widget

As far as the error, some jerk library or module from the latest build is making it, i havent isolated it yet, but more people are reporting it. I can take that issue over here to the project for the windows installer: https://github.com/pothosware/PothosSDR/issues/56

closed 2018-07-14 by guruofquality