pothosware/SoapySDR #182: Accessing SDR by serial vs driver name and index

Heshyo, 2018-08-17:
When accessing an SDR by driver and index, if I pass the wrong arguments, `make` returns `NULL`
```
SoapySDRKwargs args = {};
SoapySDRKwargs_set(&args, "driver", driverName);
SoapySDRKwargs_set(&args, "index", index);
sdr = SoapySDRDevice_make(&args);
SoapySDRKwargs_clear(&args);
if (sdr == NULL) ...
```

But if I access it by serial number, I never get `NULL`
```
SoapySDRKwargs args = {};
SoapySDRKwargs_set(&args, "driver", driverName);
SoapySDRKwargs_set(&args, "index", index);
sdr = SoapySDRDevice_make(&args);
SoapySDRKwargs_clear(&args);
if (sdr == NULL) // It's NEVER NULL!
```

As a hack I get the hardware info of the SDR and check if it's empty or not
```
SoapySDRKwargs sdrInfo = SoapySDRDevice_getHardwareInfo(sdr);
bool exist = sdrInfo.size != 0;
SoapySDRKwargs_clear(&sdrInfo);
```

Is there a reason for `make` to not return `NULL` when passing a serial, even if the serial is incorrect or the SDR is not plugged?

Heshyo, 2018-08-17:
Funny thing: I have a USRP B210. If I try to connect to it passing the proper serial number while the firmware and/or gateware files are not already loaded nor available, `make` returns `NULL`.

guruofquality, 2018-08-17:
Then it could be a UHD bug, do you see the same issue if you use uhd_usrp_probe without and without proper serial numbers?

Heshyo, 2018-08-18:
Here's what I get using uhd_find_devices and uhd_usrp_probe:

```
uhd_find_devices.exe --args "serial=12345"
No UHD Devices Found

uhd_usrp_probe.exe --args "serial=12345"
Error: LookupError: KeyError: No devices found for ----->
Device Address:
    serial: 12345
```

guruofquality, 2018-08-19:
I see what you mean, I think the factory may try to instantiate the driver without any arguments when the enumeration has zero results. And the behaviour we really want is an error. I will look into this.

If you need a work around in the mean time, try using the result from Device::enumerate() and passing that into Device::make(). enumerate will return 0 length results if you specify a serial thats not there.

Heshyo, 2018-08-20:
Thanks, for now I just make sure the HW info isn't empty. It seems reliable enough for my use case

```
SoapySDRKwargs sdrInfo = SoapySDRDevice_getHardwareInfo(sdr);
bool exist = sdrInfo.size != 0;
SoapySDRKwargs_clear(&sdrInfo);
```