pothosware/SoapySDR #185: Support for the SignalHound USB devices

Toontje, 2018-09-03:
Have you ever thought about adding support for something like https://signalhound.com/products/usb-sa44b/

guruofquality, 2018-09-03:
FWIW I have some experience with this API. The specan looks like it can be put into a raw sample mode from what I recall. The driver can also yield low power fft samples and work with a tracker generator to perform sweeps of large ranges, also yielding log power fft samples. I think there are some interesting use cases here.

Toontje, 2018-09-03:
As with the RF-Explorer. I -might- be able to get my hands on one (not
confirmed yet) and if I do I can help testing.

Toontje, 2018-11-04:
Unfortunately i was not able to get my hands on the signalHound, so i will not be able to help you out here. I do have the RF-Explorer, though.

joshspivey, 2020-03-02:
@guruofquality I got a SA44b and a bb60C. I defiantly want this it would help me out allot since I cant use slow devices with low RBW for TSCM work. Have you started working on this? right now I am integrating this API into a tensor flow signal intelligence app and I would love to add soapy to it. If you can integrate signal hound into to soapy I can add it to my app.

guruofquality, 2020-03-07:
I have no done any work on this. There is a driver guide to help get started: https://github.com/pothosware/SoapySDR/wiki/DriverGuide

ncorgan, 2022-12-26:
Looks like SignalHound themselves have a BB60C module: https://github.com/SignalHound/soapy-bb60

Since then, they've put up GNU Radio 3.9 blocks for VSG60 and the SM series.