pothosware/SoapySDR #186: make a release soon

ZeroChaos-, 2018-09-04:
just a thought, been a while

Toontje, 2018-10-14:
Please think about SdrDx integration as well?

guruofquality, 2018-10-25:
The release is tagged and for SoapyRemote as well.

ZeroChaos-, 2018-12-06:
bump, progress seems to have slowed a bit on this request.  I'd love to have this update in my livecd before the New Year :-)

guruofquality, 2018-12-06:
All thats left is updating the PPA and tagging a few hardware support projects (if they merit it). Is there any project in particular you are looking for a new tag, etc?

ZeroChaos-, 2018-12-06:
looks like soapybladerf, soapyplutosdr, soapyrtlsdr, soapysdrplay and to a lesser degree soapyuhd could all use an updated tag.  This is just based on my weak assessment that the changes since the last tag look meaningful in some way.

guruofquality, 2018-12-21:
@ZeroChaos- I think everything has a tag now. Let me know if you need something else.

For anyone following, for the PPA I'm coordinating that on myriadrf FWIW: https://discourse.myriadrf.org/t/updating-the-ppa-to-soapysdr-0-7/3892

guruofquality, 2018-12-24:
Final issue: updating the PPAs https://discourse.myriadrf.org/t/updating-the-ppa-to-soapysdr-0-7/3892

SoapySDR is up, working on other projects that depend on it now. Thanks!

closed 2018-12-24 by guruofquality