pothosware/SoapySDR #187: Hepl needed with SoapySDRServer

Toontje, 2018-09-09:
I ran SoapySDRServer --bind and connected to the port shown in the console using SdrDx.

SdrDx keeps connecting and disconnecting and this is what i get on the console:

```
SoapyServerListener::accept([::ffff:127.0.0.1]:49876)
SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv() FAIL: header word
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:127.0.0.1]:49877)
SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv() FAIL: header word
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:127.0.0.1]:49878)
SoapyServerListener::handlerLoop() FAIL: SoapyRPCUnpacker::recv() FAIL: header word
```

Wrong protocol?

guruofquality, 2018-09-09:
Can you run locally without soapyremote and see if there are any crashes. Its probably just some segfault or improper shutdown - so the threads in the server just report the error and close the socket.

Toontje, 2018-09-09:
The software doesn't support the SDRPlay natively, so i was hoping it would
work with SDR Remote. Assuming that there is a common protocol for SDR
devices. Wrong assumption?

Toontje, 2018-09-09:
How would i test that? SdrDx doesn't natively support the SDRPlay so i was hoping that remote would work. Assuming that there would be a common SDR protocol. Wrong assumption?

guruofquality, 2018-09-09:
>  Assuming that there is a common protocol for SDR

lol. come again?

so SdrDx isnt linked against soapysdr, you are just giving the SdrDx the ip address and port of the SoapyRemote server and its speaking a different protocol, and SoapyRemote just "hangs up". Thats whats happening?

BTW if thats the case, whatever the SdrDx protocol may be, a server like SoapyRemote that implements this protocol and calls into soapysdr would be great. Theres other suggestions like that, spyserver for example: https://github.com/pothosware/SoapySDR/issues/156

Toontje, 2018-09-10:
Yeah, i guess that's what's happening. What is it that needs to happen for you (or whoever on the Soapy side) and Ben (on the SdrDx side) talk to each other and see if it's a big effort to get Soapy support into SdrDx?

guruofquality, 2018-09-10:
Well either SdrDx should link against SoapySDR and call into its API to talk to SDR devices and to stream samples just like gr-osmosdr or cubicsdr does. -- OR -- someone should implement a server that mimics whatever this network protocol is that you were trying to use and talk to SoapySDR API. Any idea what the protocol is called? And honestly both options may have some utility.

Toontje, 2018-09-10:
Check mail.

guruofquality, 2018-09-12:
in any case, see https://github.com/pothosware/SoapySDR/issues/32

closed 2018-09-12 by guruofquality

ncorgan, 2022-02-27:
https://fyngyrz.com/?p=915

> All you need to do with regard to SdrDx is tell me what device name you will use with the server (it currently reports in as "RTLSVR", you need to pick a name for the device you’re supporting, and tell me what it is.)

A GPLv2 server exists to link a device to SdrRx; it wouldn't be difficult to port. All that would be necessary is to give the developer, Ben, a string to associate a SoapySDR server with it. Assuming he's still maintaining this, this is relatively simple.

zuckschwerdt, 2022-02-27:
It looks like the SdrDx protocol is a control channel with a sparse set of commands and a UDP broadcast of SDR data. Workable but it misses many nice features of SDRs. Also framing SDR data into UDP is a bottleneck and not a good abstraction.
It certainly looks like a nice UI but the project isn't very active. I'd say SDRangel is a match for features, or SigDigger for a fresh and simpler UI, and then with the classics of Gqrx and CubicSDR, maybe SDR# or things like the new SDRPlusPlus, there is a broad choice of alternatives.
I'd rather like to see attention on on #156.

ncorgan, 2022-02-27:
SDRPlusPlus and SDRangel already support Soapy. I hadn't heard of SigDigger, but apparently it does too.

I dove a bit into the SdrDx client, and I agree on the limitations. I'm already working on a SpyServer device and SDR# plugin (among some others) when I can, so that should widen support significantly.

ncorgan, 2022-02-27:
I have a few things I haven't pushed. With the possible exception of the SpyServer device, which is in the testing phase, these projects are too early-on to be usable, but can put them up (at least in private repos) for now with that disclaimer.