pothosware/SoapySDR #188: SoapyRTLSDR module installation fails with librtlsdrSupport.so failed

zanco, 2018-11-04:
Hi, trying to make rtty decoder run on Pi 2B. Needed to update from Jessie to Stretch. decoder runs with SoapySDR so I installed SoapySDR which requires the RTLSDR module (that's the receiver)

Running cmake ../ goes without errors.
Running make ends with Collert2 Error:

''collect2: error: ld returned 1 exit status
CMakeFiles/rtlsdrSupport.dir/build.make:174: recipe for target 'librtlsdrSupport.so' failed
make[2]: *** [librtlsdrSupport.so] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/rtlsdrSupport.dir/all' failed
make[1]: *** [CMakeFiles/rtlsdrSupport.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
''

rtl-test works without errors. 

What could be the problem ? I do not get a clue from the error message.

Thanks,
Ben

guruofquality, 2018-11-04:
> undefined reference to SoapySDR::Registry::Registry(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<std::map<std::__cxx11::basic_string<char, std::char_traits<char>, 

Do you have old development files for SoapySDR on the system in /usr and /usr/local? Maybe its looking at mismatched copies of the library and headers. Or if these many of those errors, it might not be linking correctly, not sure.

Can you make sure you have a clean development environment with just the one copy of SoapySDR development files, and retry? If that fails can you attach make VERRBOSE=1, it will show the linker flags and include flags, etc

zanco, 2018-11-04:
Thanks for the reply. There are a lot of those errors. I did run make -d with output to a text file, I was unable to get --verbose or --VERBOSE-1 working. Could be my mistake ? 

https://pastebin.com/kXN3HLLN

Has the output of make -d

This Pi has had no previous installation of SoapySDR. Only Pi in the Sky software has been used so far. The SoapySDR itself did build with no errors. Building the rtlsdr module fails.

zanco, 2018-11-05:
sudo make uninstall SoapySDR
sudo rm -r Soapy*

cloned  SoapySDR again, built succesfully
cloned SoapyRTLSDR again, built succesfully.

> Lib Version: v0.7.0-g6d21756e
API Version: v0.7.0
ABI Version: v0.7
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/local/lib/SoapySDR/modules0.7/librtlsdrSupport.so (0.3.0-f409a56)
Available factories... rtlsdr

Sometimes things need a fresh start. Thanks for your input !

Ben

closed 2018-11-06 by zanco