pothosware/SoapySDR #189: rx_tools on RSP1A

Toontje, 2018-11-04:
I used to use rx_power from rx_tools on the RSP1. On the RSP1A i get the following errors:
```
Error: bad retune at 88500000 Hz (4 of 3 attempts), r=-5, flags=0 (try increasing -S or -R).
Error: reading stream -5
```
Any idea where that comes from? Rx_tools is supposed to be using SoapySDR as back-end.

starvald, 2018-11-04:
Have an rsp1a too not seeing a problem here.

```
[lee@Plasma ~]$ rx_power -f 88M:108M:125k fm_stations.csv
Number of frequency hops: 8
Dongle bandwidth: 2500000Hz
Downsampling by: 1x
Cropping by: 0.00%
Total FFT bins: 256
Logged FFT bins: 256
FFT bin size: 78125.00Hz
Buffer size: 16384 bytes (3.28ms)
Reporting every 10 seconds
Found 1 device(s):
  0: driver=sdrplay, label=SDRplay Dev0 RSP1A xxxxxxxxx, 

verbose_device_search((null))
Using device 1804043895: mir_sdr_api_version=2.130000 mir_sdr_hw_version=255 
Found 1 antenna(s): RX 
Found 2 gain(s): IFGR RFGR 
Found 2 frequencies: RF CORR 
Found 13 sample rates: 250000 500000 1000000 2000000 2048000 3000000 4000000 5000000 6000000 7000000 8000000 9000000 10000000 
Found 8 bandwidths: 200000 300000 600000 1536000 5000000 6000000 7000000 8000000 
[INFO] Using format CS16.
```

unixpunk, 2018-11-04:
Not sure if you've tried, but I had to recompile SoapySDRPlay module after upgrading the SDRPlay API.  rx_tools works fine for my RSP1a after that.  HTH

guruofquality, 2018-11-04:
Null is a fall through option for SoapySDR::Device::make() when theres no args and no devices found.

Seems strange since SoapySDRPlay seems to instantiate just fine.

Whats the output for `SoapySDRUtil --find` and `SoapySDRUtil --make` without any arguments?

And does it change anything to pass arguments to rx_power like `driver=sdrplay`

Toontje, 2018-11-04:
```
tmachielsen@Laptop:~ $ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  driver = sdrplay
  label = SDRplay Dev0 RSP1A 1807058E93

tmachielsen@Laptop:~ $ SoapySDRUtil --make
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Make device
  driver=SDRplay
  hardware=1807058E93
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=255
```

Toontje, 2018-11-08:
That is correct:

```
tmachielsen@Laptop:~ $ find /usr -name libSoapy* 2>/dev/null
/usr/local/lib/libSoapySDR.0.7.dylib
/usr/local/lib/libSoapySDR.0.7.0.dylib
/usr/local/lib/libSoapySDR.dylib
/usr/local/Cellar/soapysdr/0.6.1/lib/libSoapySDR.0.6.1.dylib
/usr/local/Cellar/soapysdr/0.6.1/lib/libSoapySDR.dylib
/usr/local/Cellar/soapysdr/0.6.1/lib/libSoapySDR.0.6.dylib
/usr/local/Cellar/pothossoapy/0.5.0/lib/Pothos/modules0.6/soapy/libSoapySupport.so
/usr/local/Cellar/pothossoapy/0.5.0/lib/Pothos/modules0.6/soapy/libSoapyUtilityBlocks.so
```

guruofquality, 2018-11-09:
Well I guess thats the answer then. So I think if you clean things, brew reinstall, and or install rx_tools however, it probably should be ok.

FYI, I did update brew for the latest release tag also.

closed 2018-11-09 by guruofquality

Toontje, 2018-11-09:
We are not quite there yet. I still need some help:

```
tmachielsen@Laptop[±|master U:1 ?:14 ✗]:~/rx_tools $ ./rx_sdr
dyld: Library not loaded: /usr/local/opt/soapysdr/lib/libSoapySDR.0.6.dylib
  Referenced from: /Users/tmachielsen/rx_tools/./rx_sdr
  Reason: image not found
Abort trap: 6
```
Ok, so libSoapy 0.6 doesn't exist anymore. Let's reinstall it.

```
tmachielsen@Laptop[±|master U:1 ?:14 ✗]:~/rx_tools $ brew install soapysdr
Warning: pothosware/pothos/soapysdr 0.7.0 is already installed, it's just not linked
You can use `brew link soapysdr` to link this version.
tmachielsen@Laptop[±|master U:1 ?:14 ✗]:~/rx_tools $ brew link soapysdr
Linking /usr/local/Cellar/soapysdr/0.7.0...
Error: Could not symlink bin/SoapySDRUtil
Target /usr/local/bin/SoapySDRUtil
already exists. You may want to remove it:
  rm '/usr/local/bin/SoapySDRUtil'

To force the link and overwrite all conflicting files:
  brew link --overwrite soapysdr

To list all files that would be deleted:
  brew link --overwrite --dry-run soapysdr
tmachielsen@Laptop[±|master U:1 ?:14 ✗]:~/rx_tools $ brew link --overwrite soapysdr
Linking /usr/local/Cellar/soapysdr/0.7.0... 10 symlinks created
```
Let's see if it works now.
```
tmachielsen@Laptop[±|master U:1 ?:14 ✗]:~/rx_tools $ ./rx_sdr
dyld: Library not loaded: /usr/local/opt/soapysdr/lib/libSoapySDR.0.6.dylib
  Referenced from: /Users/tmachielsen/rx_tools/./rx_sdr
  Reason: image not found
Abort trap: 6
```
Hmmm, 0.7 in /usr/local/Cellar but rx_tools still wants to use the one in /usr/local/lib. Let's clean the makefiles and compile rx_toosl again.

```
tmachielsen@Laptop[±|master ?:14 ✗]:~/rx_tools $ rm -rf CMakeFiles/ CMake
tmachielsen@Laptop[±|master U:1 ?:13 ✗]:~/rx_tools $ cmake .
-- The C compiler identification is AppleClang 10.0.0.10001145
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Looking for pthread_create
-- Looking for pthread_create - found
-- THREADS_PTHREADS_INCLUDE_DIR:
-- CMAKE_THREAD_LIBS_INIT:
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/tmachielsen/rx_tools
tmachielsen@Laptop[±|master U:1 ?:14 ✗]:~/rx_tools $ make
Scanning dependencies of target common
[ 12%] Building C object CMakeFiles/common.dir/src/convenience/convenience.c.o
[ 25%] Linking C static library libcommon.a
[ 25%] Built target common
Scanning dependencies of target rx_sdr
[ 37%] Building C object CMakeFiles/rx_sdr.dir/src/rtl_sdr.c.o
[ 50%] Linking C executable rx_sdr
[ 50%] Built target rx_sdr
Scanning dependencies of target rx_power
[ 62%] Building C object CMakeFiles/rx_power.dir/src/rtl_power.c.o
[ 75%] Linking C executable rx_power
[ 75%] Built target rx_power
Scanning dependencies of target rx_fm
[ 87%] Building C object CMakeFiles/rx_fm.dir/src/rtl_fm.c.o
[100%] Linking C executable rx_fm
[100%] Built target rx_fm
tmachielsen@Laptop[±|master U:1 ?:14 ✗]:~/rx_tools $ ./rx_sdr
dyld: Library not loaded: /usr/local/opt/soapysdr/lib/libSoapySDR.0.6.dylib
  Referenced from: /Users/tmachielsen/rx_tools/./rx_sdr
  Reason: image not found
Abort trap: 6
```
Still no progress... How do i decide where rx_tools finds the soapy libraries?

zuckschwerdt, 2018-11-09:
`otool -L rx_sdr` will help you to discover the linkage.

zuckschwerdt, 2018-11-09:
Oh sorry, I guess you already know you linked your rx_tools to that `/usr/local/opt/soapysdr/lib/libSoapySDR.0.6.dylib` path, which doesn't seem to be around?

If you want to force a include and/or lib use (out of tree build like this is recommended):
```
mkdir build ; cd build
cmake \
 -DSOAPY_SDR_INCLUDE_DIR=/opt/local/include \
 -DSOAPY_SDR_LIBRARY=/opt/local/lib/libSoapySDR.dylib \
.. && make
```

Toontje, 2018-11-09:
Did an uninstall of everything soapy and pothos related. :-(
Did a brew cleanup.
Did a reinstall of only soapysdr using homebrew.
Did a rebuild of SoapySDRplay (not in homebrew (yet?)).
Did a rebuild of rx_tools.

All working.

Freaking dependecies..... :-)

zuckschwerdt, 2018-11-09:
For reference, you likely removed the lib and include but missed the pkg-config or cmake modules. In your case:

    /usr/local/opt/soapysdr/lib/pkgconfig/SoapySDR.pc
    /usr/local/opt/soapysdr/share/cmake/SoapySDR/SoapySDRConfig.cmake