pothosware/SoapySDR #190: Feature Request: Automatic IQ balance correction

nsdoukos, 2018-11-05:
I have noticed that some devices like usrp b210 and e310 support automatic IQ balance correction. 
Right now Soapy supports only setting IQ balance manually which is something the devices mentioned don't support so IQ balance correction functionality cannot be used with SoapySDR.

guruofquality, 2018-11-05:
> I have noticed that some devices like usrp b210 and e310 support automatic IQ balance correction.

Cool, do you know the API hooks for this? I would like to look at it. I'm on board with adding this as long as there is an API for it. I remember there was a calibration utility, so if you call the calibration utility, the corrections are used when the LO is re-turned (or something like that).

> the devices mentioned don't support so IQ balance correction

Is this a recent change in UHD or the FPGA images for these devices? I remember putting in the same IQ correction block into b210 and e310 FPGA, same as n210. Could be wrong...

In any case, the SoapySDR hooks call into set_tx/rx_iq_balance() and while you don't have to use the manual API, these should be the same calls used by the calibration utility.

```
void setIQBalance(const int dir, const size_t channel, const std::complex<double> &balance)
    {
        if (dir == SOAPY_SDR_TX) _dev->set_tx_iq_balance(balance, channel);
        if (dir == SOAPY_SDR_RX) _dev->set_rx_iq_balance(balance, channel);
    }
```

So at least that stays regardless :-)

nsdoukos, 2018-11-06:
Hello again, sorry for the delay. 

> 
> Cool, do you know the API hooks for this? I would like to look at it. I'm on board with adding this as long as there is an API for it. I remember there was a calibration utility, so if you call the calibration utility, the corrections are used when the LO is re-turned (or something like that).

virtual void set_rx_iq_balance(const bool enb, size_t chan) = 0;

is the API function provided by the uhd.

> 
> > the devices mentioned don't support so IQ balance correction
> 
> Is this a recent change in UHD or the FPGA images for these devices? I remember putting in the same IQ correction block into b210 and e310 FPGA, same as n210. Could be wrong...

I don't know if there is a change. Maybe it didn't print the following warning :
[WARNING] Setting IQ balance is not possible on this device.

> void setIQBalance(const int dir, const size_t channel, const std::complex<double> &balance)
>     {
>         if (dir == SOAPY_SDR_TX) _dev->set_tx_iq_balance(balance, channel);
>         if (dir == SOAPY_SDR_RX) _dev->set_rx_iq_balance(balance, channel);
>     }
> ```
> 
> So at least that stays regardless :-)

Yes of course, other usrp devices should support it.

guruofquality, 2019-11-19:
API is merged, remaining TODO:
https://github.com/pothosware/SoapySDR/commit/67abec9fe7ba84b4d069951c6e7f8c45f11e8335

* [ ] SoapyRemote support for new call
* [ ] SoapyUHD support
* [ ] UHDSoapy support

closed 2022-11-24 by ncorgan