pothosware/SoapySDR #192: Slower setFrequencyComponent with 2018-11-04 version

KarlL2, 2018-11-08:
Hi, sorry if this is not the right place, I'm not sure where to ask this.

I have a Lime and I need to transmit at different frequencies. I use setFrequencyComponent with BB to shift the transmit frequency using the NCO.

With a previous version of Pothos (around April 2018 I think), the call would take around 2.2ms while now it takes 4.4ms. Any idea what could be wrong?

Thanks.

guruofquality, 2018-11-08:
> Hi, sorry if this is not the right place, I'm not sure where to ask this.

I don't know enough to say if its limesuite driver perhaps, or something in PothosCore.  But I'm doubtful that there is anything that I can fix in this exact project per say.

> I have a Lime and I need to transmit at different frequencies. I use setFrequencyComponent with BB to shift the transmit frequency using the NCO.
> With a previous version of Pothos (around April 2018 I think), the call would take around 2.2ms while now it takes 4.4ms. Any idea what could be wrong?

So I take it that this is the windows installer. Obviously a lot changed since april here and there. Driver updates, releases of things, and minor and major changes. How are you calling setFrequencyComponent()? -- an API call, python, C++ or is this from the GUI?

So one possibility is its some change in **PothosCore**. For this latest installer I changed to the master branch of PothosCore and various blocks. There were some threading changes and they can be suspect. One thing you can try is to install the previous installer which should have been on version 0.6, same as april I believe. See if that isolates it.

More likely, I think that its just a change in **limesuite**. If thats the case, you might want to try different installers and see what version is reported by LimeUtil --info (I think) to narrow it down. Or if you can get a development environment, git bisect and see which commit or release introduces the problem.

* setFrequencyComponent with BB ultimately calls lms7Device->SetNCOFreq() (see https://github.com/myriadrf/LimeSuite/blob/master/src/API/lms7_device.h) So you can also make a simpler utility to time just setting that.

Anyway, thats just some tips on bisecting the problem. If there is some regression in limesuite driver, the LimeSuite team needs to know: https://github.com/myriadrf/LimeSuite/issues

Or if its in PothosCore with calling slots on the master branch, I need to know. And it should be easier to pin that down specifically with some testing if thats the case: https://github.com/pothosware/PothosCore/issues

rjonaitis, 2018-11-08:
Most likely caused by LimeSuite change: https://github.com/myriadrf/LimeSuite/commit/c02f4358b6eb3a83f333e8951419f34b18667dde#diff-d2304d4575d79f2082317421ac086e5bL888

KarlL2, 2018-11-09:
Thanks. So I checked different versions of Pothos:

- PothosSDR-2018.04.08-vc14-x64.exe gives me the 2.2ms
- PothosSDR-2018.04.26-vc14-x64.exe gives me the 4.4ms

I don't know if it correspond to the change mentioned by @rjonaitis commited on February 2. Is there a way to know what commit correspond to what PothosSDR version?

guruofquality, 2018-11-09:
LimeUtil should be able to tell you the version. That commit may not have been merged into master until a later date so the exact date of the commit may not specifically help, but whether or not the code change is in there from the git hash in LimeUtil should work.

@rjonaitis I didnt really follow the change you mentioned, I guess in my opinion setting the NCO should always remain as fast as possible for those who might use it to frequency hop. But I'm not sure what was the issue.

Anyway, I'm going to close this, because there isnt anything I can fix in SoapySDR itself.

closed 2018-11-09 by guruofquality