pothosware/SoapySDR #193: Soapy CMAKE Modifies Project Build Flags

bhilburn, 2018-11-09:
As seen [here](https://github.com/pothosware/SoapySDR/blob/66306d24e07d2cd67f292286f3f460734700c234/cmake/Modules/SoapySDRConfig.cmake#L70), Soapy's CMAKE code, which will be invoked by `find_package` calls in projects that include Soapy, actually modifies the CMAKE flags of the including project.

Changing the build flags of an including codebase is problematic in that it (a) breaks encapsulation and (b) is unexpected from a dependency. If these flags are _required_ to include a pre-built dependency for some reason, I think CMAKE should issue errors / warnings rather than silently changing build flags.

guruofquality, 2018-11-09:
Yes, I agree. CMake is a lot better at handling stuff like this than it used to be. There an open issue to switch to modern cmake, so all of this would just be set per target and not pollute the global settings. And it probably only needs to set C++11 for that matter or some defaults enabled with an option on the add module macro.

* https://github.com/pothosware/SoapySDR/issues/152

Sadly I will have to hear about this breaking compatibility on ubuntu precise LTS or whatever :-)

In the meantime, if you can somehow stick the find_project(SoapySDR CONFIG) into a subdirectory, it should help.

guruofquality, 2018-11-11:
@bhilburn So Im going to rewrite things for modern cmake, but since no one has actually complained before im going to see if I can get away with more moderate changes for a sooner rather than later release tag.

* do you need this fix in a release tag?
* if I fixed find_package() so it didn't pollute, but SOAPY_SDR_MODULE_UTIL() set the same global settings, would that be a problem?
  * (this would be a yes if you have support modules or said modules are being setup in a global namespace with other project stuff below it)
* Is there a specific platform (eg linux) or flag thats exacerbating the issue?

bhilburn, 2018-11-12:
Awesome!

* No, I don't need a release tag.
* Intra-module settings within Soapy are fine (possible I'm mis-understanding your ask, here, though)
* Currently, forcing explicit exporting of symbols is an issue.