pothosware/SoapySDR #194: WIKI Build Gude Update.

tdbone1, 2018-11-12:
i have successfully built SoapySdr and Pothos LimeSuite from git using Kali-linux i386.
i was just wondering what do you do when git has updates and you want to update your binaries you already have installed.
do you have to 'make clean' and remove any files manually
it would be nice to add how to properly updating from git after you have successfully built (according to the wiki build guide.)

guruofquality, 2018-11-12:
For 99% of these projects, updating is basically
```
cd source_directory; git pull
cd build_directory; make && sudo make install
#and for debian based distros, they make need sudo ldconfig if the library bumped so numbes
```

guruofquality, 2018-12-10:
I added a note in the unix build instructions: https://github.com/pothosware/SoapySDR/wiki/BuildGuide#unix-instructions

closed 2018-12-10 by guruofquality