pothosware/SoapySDR #195: Use specific device methods / fields

KarlL2, 2018-11-14:
Hi Josh,

I was wondering if there's any way to access the real device and not just the abstracted device?

I have a program that uses SoapySDR and it works well. I can use it with different SDRs. But in a specific part of my code, for a specific SDR, I would want to be able to access the "real" device directly and call methods specific to it. Is there any existing thing for this?

More specifically I would like to do:

    SoapySDRDevice* soapySdr = ... ;
    auto lms7_device = (lime::LMS7_Device*)soapyLMS7->getRealDevice();
    lms7_device->callWhateverSpecificFunction();

----

Actually I tried recompiling SoapyLMS7 on Windows to make the lms7Device public instead of private, so I can do:

    SoapyLMS7* soapyLMS7 = (SoapyLMS7*)soapySdr;
    lime::LMS7_Device* lms7_device = soapyLMS7->lms7Device;

but then calling anything on lms7_device throws an exception. I wasn't able to compile with "Visual Studio 14 2015 Win64", instead I compiled it with "Visual Studio 15 2017 Win64". I don't know if it could be the issue.

zuckschwerdt, 2018-11-14:
@KarlL2 maybe you want to extend the `writeSetting()` and `readSetting()` in the SoapyLMS7 driver to implement that call. If it's useful an generic enough it might be accepted as enhancement.
See e.g. the [digital stage AGC in SoapyRTLSDR](https://github.com/pothosware/SoapyRTLSDR/blob/master/Settings.cpp#L585) for reference.

KarlL2, 2018-11-15:
Thanks for the idea @zuckschwerdt . I thought it'd be better to get the real device as you can then call anything you want, but you have to include more headers.

Instead, while looking at the `writeSetting()`, I found the `SoapySDRDevice_writeRegister` which seems to be exactly what I need. For my use case I just need to set a few registers, so I should be able to do it without modifying Soapy. Thanks again.

closed 2018-11-15 by KarlL2

KarlL2, 2018-11-16:
So eventually I was able to do what I wanted by only using `SoapySDRDevice_writeRegister` and `SoapySDRDevice_readRegister`.

On the down side I had to duplicate some internal methods, and I was lucky that by not using the internal cache I could end up in the right `if / else` statements.

KarlL2, 2018-11-18:
Thanks Josh. For context that was my question [Frequency hopping transmit](https://discourse.myriadrf.org/t/frequency-hopping-transmit/3622?u=karll) where I wanted to be able to go *fast* from one frequency to another.

With an older build, it took 2.2ms while with more recent builds it takes 4.4ms just to modify the NCO ( https://github.com/pothosware/SoapySDR/issues/192 ). Now, after calling the *normal* function once to set things up, and getting a reference clock value, I can simply modify 2 registers through the Soapy interface, and it takes around 0.400ms.

I'm not too familiar with C++, but what you describe is similar to reflection in C#. That could be an interesting feature indeed.