pothosware/SoapySDR #199: Unable to install soapysdr-lms7 due to versioning issue

phansel, 2018-12-14:
`paul@comp22:~$ sudo apt install soapysdr-lms7 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 soapysdr-lms7 : Depends: liblimesuite16.8-1 (= 16.8.1.819.792.4165104-myriadrf1~xenial) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
`

However, after install liblimesuite-dev, apt rejects it on a version basis:

liblimesuite-dev is already the newest version (18.10.0-myriadrf1~xenial).

Linux comp22 4.15.0-42-generic #45~16.04.1-Ubuntu SMP Mon Nov 19 13:02:27 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

guruofquality, 2018-12-14:
Dont install whatever that is. Maybe very ancient. The package is called soapysdr-module-lms7 (see debian/control in LimeSuite)

Also I need to update the PPAs for the latest SoapySDR release. Thats a big update, and I have to do it all at once. But I think everything up there now is build together and compatible. Let me know if yo run into that! thanks

guruofquality, 2018-12-17:
So, anyway, please edit any wiki that says to install soapysdr-lms7, I hope that helped!

closed 2018-12-17 by guruofquality