pothosware/SoapySDR #200: List of SDR software with Soapy support

ast, 2019-01-02:
It would be great to extend the list of SDR software that supports Soapy at the bottom of the bottom of the README.md. GQRX has good support and CubicSDR is supposed to work, but I never got it working properly... 

We are building a Raspberry Pi based SDR transciever (tujasdr.com) and are using Soapy as our "native" interface so I'm very interested to learn about all available software..

In particular there seems to be a lack of software for Windows and software with good support for tx...

guruofquality, 2019-01-06:
I concur, I tend to add things here and there that I found out about, but I would be great to see folks more proactive about sharing their projects. Like the other day or so, I accidentally found out that someone was working on gr-soapy gnuradio blocks. Thats good to list here for others to see :-)

* Some examples listed here: https://github.com/pothosware/SoapySDR/wiki#example-platforms
* And the list on the sidebar
```
* `Pothos SDR`_
* `Cubic SDR`_
* `Rx Tools`_
* `LuaRadio`_
* `GNURadio blocks`_
* `Osmocom blocks`_
* `QSpectrumAnalyzer`_
```

@ast PS -- https://github.com/pothosware/SoapyFCDPP/issues/3

nmaster2042, 2019-01-12:
Hi.

There is some other softwares that support SoapySDR.
Here is what I found:

- rtl_433: https://github.com/merbanan/rtl_433
- welle.io: https://www.welle.io/

nmaster2042, 2019-01-12:
I forgot sdrangel SDR software https://github.com/f4exb/sdrangel
It now has SoapySDR support.

starvald, 2019-01-21:
rtl_airband https://github.com/szpajder/RTLSDR-Airband

guruofquality, 2019-01-21:
Thanks, I have been updating the wiki and sidebar.

PS Wow, it really is a lot of receive-centric apps :-)

nmaster2042, 2019-02-16:
I found another SDR program that now supports SoapySDR:

LinHPSDR https://github.com/g0orx/linhpsdr

It seems to support transmit and receive but I didn't tryed it.

ogre, 2019-02-16:
https://github.com/ogre/habdec

guruofquality, 2019-03-09:
A transmitter application: https://github.com/fsphil/hacktv

guruofquality, 2019-03-14:
https://github.com/Opendigitalradio/ODR-DabMod

nmaster2042, 2019-06-21:
A new receive app get added SoapySDR support: QUISK http://james.ahlstrom.name/quisk/

nmaster2042, 2019-11-05:
A signal analyzer app that support SoapySDR: SigDigger https://github.com/BatchDrake/SigDigger

fransvanberckel, 2020-10-10:
It's optional, but openwebrx does SoapySDR as well ... https://github.com/jketterl/openwebrx/wiki/Manual-Package-installation-(including-digital-voice)

kantooon, 2020-10-10:
https://github.com/qradiolink/qradiolink very experimental but supports transmitting, at least with SoapyPlutoSDR and SoapyLMS7.

nmaster2042, 2022-01-03:
A new SDR software with SoapySDR support: SDR++: https://github.com/AlexandreRouma/SDRPlusPlus

nmaster2042, 2022-10-24:
A DAB/DAB+ new software with SoapySDR support: AbracaDABra https://github.com/KejPi/AbracaDABra

luzpaz, 2023-11-17:
Agreed. Can a PR be made for this ?

ka9q, 2026-09-06:
I just created ka9q/SoapyKA9Q.git, a SoapySDR client module for applications to use a channel from ka9q-radio (ka9q/ka9q-radio.git) as a virtual front end.

zuckschwerdt, 2026-09-07:
https://github.com/ka9q/SoapyKA9Q has been added to the wiki sidebar.

kantooon, 2026-09-07:
You can add MMDVM-IQ: https://github.com/g4klx/MMDVM-IQ and MMDVM-Multi: https://github.com/g4klx/MMDVM-Multi to the list as well.

nmaster2042, 2026-09-29:
There is a new promising  multiplateform SDR app Called SDROxide withe SoapySDR support: https://github.com/dividebysandwich/sdroxide