pothosware/SoapySDR #205: OpenLTE with soapyUHD

Dastgsm, 2019-01-09:
I'm trying openlte with soapyUHD and i have this error! 
UHD Error: Tx Calibration: MCU error 5 (Loopback signal weak: not connected/insufficient gain?) – Rx calibration finished  UHD Warning: popping from TX, samples popped 540/1020.

any idea to fix it ?

Dastgsm, 2019-01-09:
Note that i use limesdrmini.

Dastgsm, 2019-01-10:
so to have best IQ balance or DC , the Tx send calibration to the Rx and SoapySDR try to see how to have a best synchronisation with ?

guruofquality, 2019-01-12:
The LimeSDR calls calibration when the stream is activated: https://github.com/myriadrf/LimeSuite/blob/master/SoapyLMS7/Streaming.cpp#L203

If you think something is wrong or you have a support question, you need to ask the maintainers:

* https://github.com/myriadrf/LimeSuite
* https://discourse.myriadrf.org/

closed 2019-01-12 by guruofquality