pothosware/SoapySDR #213: Crash on MacOSX with rx_power

dropdrop20, 2019-03-01:
yyyyyy@XXXXXXXX ~/Projects/rx_tools $sudo rx_fm -f 88500000 aaaa.wav
[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [65: No route to host]
SoapySDRDevice_make failed
Segmentation fault: 11

guruofquality, 2019-03-01:
* What hardware are you using?
* What support modules are installed?
* output of SoapySDRUtil --info
* Does the device successfully instantiate with SoapySDRUtil --make

zuckschwerdt, 2019-03-01:
Fix in https://github.com/rxseger/rx_tools/pull/73

dropdrop20, 2019-03-01:
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.1-release
API Version: v0.7.1
ABI Version: v0.7
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.7
Module found: /usr/local/lib/SoapySDR/modules0.7/libremoteSupport.so  (0.5.1-0a7cb58)
Module found: /usr/local/lib/SoapySDR/modules0.7/libsdrPlaySupport.so (0.1.0-12c3db6)
Available factories... remote, sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

dropdrop20, 2019-03-01:
Oh... The USB cable of RSP was disconnected.

Now is better:

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [65: No route to host]
Found device 0
  driver = sdrplay
  label = SDRplay Dev0 RSP1A 18070ED596

So, the crash is indeed due to rx_fm and may be on all rx_* tools

zuckschwerdt, 2019-03-01:
Can you try the fix in https://github.com/rxseger/rx_tools/pull/73 i.e. use https://github.com/zuckschwerdt/rx_tools/tree/fix-segf

dropdrop20, 2019-03-01:
Working! You can commit in master branch

guruofquality, 2019-03-01:
ty!

closed 2019-03-01 by guruofquality