pothosware/SoapySDR #218: Latest head failed to build on raspbian because missing -pthread

ast, 2019-04-05:
Had to add:

`SET(CMAKE_CXX_FLAGS -pthread)`

To CMakeLists.txt.

This bug also applies to SoapyRemote.

Compiler clang 7.0.0.

zuckschwerdt, 2019-04-05:
I don't see this on Raspbian (Debian 9.8) using v0.8.0-g665025e4. Can you provide more info, perhaps logs output?

zuckschwerdt, 2019-04-05:
Ok, I see. Clang. Can you remove the line `if(CMAKE_COMPILER_IS_GNUCXX)` and matching `endif()` in `lib/CMakeLists.txt` and confirm that works? Or maybe pull the line `target_link_libraries(SoapySDR PRIVATE -pthread)` before the `if()`.
The fix would be a check for Clang there (maybe: `OR "${CMAKE_CXX_COMPILER_ID}" MATCHES "Clang"`) -- the gcc options should work with clang too, not sure.

ast, 2019-04-05:
That was indeed the problem. I changed that line to:
`if("${CMAKE_CXX_COMPILER_ID}" STREQUAL "GNU" OR "${CMAKE_CXX_COMPILER_ID}" STREQUAL "Clang")`

according to this answer https://stackoverflow.com/questions/10046114/in-cmake-how-can-i-test-if-the-compiler-is-clang

guruofquality, 2019-04-05:
thanks, I pushed that compiler if/else change

closed 2019-04-05 by guruofquality