pothosware/SoapySDR #219: clang linking issue

ra1nb0w, 2019-04-11:
During making a port of sdrangel (MacPorts) I found this strange linking issue with SoapySDR.

```make
Undefined symbols for architecture x86_64:
"SoapySDR::loadModule(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)", referenced from:
DeviceSoapySDRScan::scan()      in devicesoapysdrscan.cpp.o
"SoapySDR::listModules[abi:cxx11]()", referenced from:
DeviceSoapySDRScan::scan()      in devicesoapysdrscan.cpp.o
"SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)", referenced from:
DeviceSoapySDR::openopenSoapySDRFromSequence(unsigned int) in devicesoapysdr.cpp.o
DeviceSoapySDRScan::scan()      in devicesoapysdrscan.cpp.o
"SoapySDR::Device::enumerate(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)", referenced from:
DeviceSoapySDRScan::scan()      in devicesoapysdrscan.cpp.o
"SoapySDR::Registry::listFindFunctions[abi:cxx11]()", referenced from:
DeviceSoapySDRScan::scan()      in devicesoapysdrscan.cpp.o
ld: symbol(s) not found for architecture x86_64
collect2: error: ld returned 1 exit status
make[2]: *** [devices/soapysdr/libsoapysdrdevice.dylib] Error 1
make[1]: *** [devices/soapysdr/CMakeFiles/soapysdrdevice.dir/all] Error 2
```
- sdrangel is built with gcc8
- SoapySDR is build with Apple clang-1001.0.46.3 (XCode 10.2)

if I build SoapySDR with gcc8 everything works as expected. I checked the cmake configuration of sdrangel and everything is correct. Moreover, I tried to enable gcc directives  of SoapySDR and use clang but the problem persist. I see that there is some changes on *CMakeLists.txt from 0.7.1 to master but both have the same issue.

Have you some idea on how to fix or where can be the problem?
thank you

guruofquality, 2019-04-11:
Can you try some flags in the top level CMakeLists.txt? I just need to know what the linker flags should be or apple clang to use c++11

Ive seen this before
```
#enable c++11 extensions for OSX
if (APPLE)
   set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x -Wc++11-extensions")
endif(APPLE)
```

Or maybe some of these
```

clang++ -std=c++11 -stdlib=libc++ input.cxx -o a.out
clang++ -std=gnu++11 -stdlib=libc++ input.cxx -o a.out
```

* reference: https://stackoverflow.com/questions/19774778/when-is-it-necessary-to-use-use-the-flag-stdlib-libstdc

ra1nb0w, 2019-04-11:
- Test 1 - enable c++11 extensions for OSX
result https://gist.github.com/ra1nb0w/78b34130f4f41bbbcd94c757bf52a97b
- Test 2 - std=c++11 -stdlib=libc++
result https://gist.github.com/ra1nb0w/5a46ef5b2d79a25ec63289244022d426
seems that it rewrite to `--stdlib=libc++`
- Test 3 - std=gnu++11 -stdlib=libc++
result https://gist.github.com/ra1nb0w/ad10246352d3a8efe36dfe7e008b7680

For all sdrangel fails to link. Tests done with 0.7.1

FYI: `--stdlib=libc++` is the default on macOS 10.14 and `-stdlib=libstdc++` is not more supported

guruofquality, 2019-04-11:
Sorry, I meant can you try to add these flags to SDRangel? As far as I know there is more than one libstdc++ and because SoapySDR has C++11, I think SDRangel needs to link against the same runtime. So it either needs one of these std= or stdlib= flags (I think). If we figure out the flag, I might be able to export it with the CMakeConfig for SoapySDR so its automatically used by SDRangel.

ra1nb0w, 2019-04-12:
The code is already built with `set (CMAKE_CXX_STANDARD 11)`
`-stdlib=` doesn't exist on gcc.
This is the sdrangel build of soapy devices https://gist.github.com/ra1nb0w/d13f243e3ee8b7fb9abf8edcf830dec3

In SoapySDR I don't understand how set `-std=gnu++11`

guruofquality, 2019-04-12:
Its strange that its just the factory functions. Another possibility is that SDRangel is just somehow not linked to SoapySDR? Can you show the output of `make VERBOSE=1` for SDRangel build That might help anyway.

ra1nb0w, 2019-04-12:
[this](https://gist.github.com/ra1nb0w/281aa85a5cbb08b60bb9371b9a5bc4ba) is the output SDRAngel with `VERBOSE=1` for SoapySDR module

guruofquality, 2019-04-13:
So I compiled SoapySDR and sdrangel with clang on a HighSierra VM and it builds fine: https://pastebin.com/C5Uu1sCH Also notably, the make VERBOSE=1 actually shows the commands with the options and linker flags. Not sure what your snippet is different. So I don't know what to do. I'm happy to add flags, but this could just be a bad install.

I also had to make a few changes to sdrangel. the freedv project has some inconsistent use of complex, C99, and C++11 feature set. So it was a little challenge to get it to compile:

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index c590030c..ad26e5e3 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -54,6 +54,13 @@ endif()
 set(QT_USE_QTOPENGL TRUE)
 set(CMAKE_AUTOMOC ON)
 
+if(APPLE AND EXISTS /usr/local/opt/qt5)
+       # Homebrew installs Qt5 (up to at least 5.9.1) in
+       # /usr/local/qt5, ensure it can be found by CMake since
+       # it is not in the default /usr/local prefix.
+       list(APPEND CMAKE_PREFIX_PATH "/usr/local/opt/qt5")
+endif()
+
 #find_package(Qt4 REQUIRED)
 find_package(Qt5Core 5.0 REQUIRED)
 find_package(Qt5Widgets 5.0 REQUIRED)
diff --git a/libfreedv/CMakeLists.txt b/libfreedv/CMakeLists.txt
index b37d8315..d3bc10ad 100644
--- a/libfreedv/CMakeLists.txt
+++ b/libfreedv/CMakeLists.txt
@@ -1,5 +1,7 @@
 project(freedv)
 
+set(CMAKE_CXX_STANDARD 11)
+
 set(freedv_SOURCES
        codec2_fft.cpp
        cohpsk.cpp
@@ -80,4 +82,4 @@ target_link_libraries(freedv
        ${CODEC2_LIBRARIES}
 )
 
-install(TARGETS freedv DESTINATION lib)
\ No newline at end of file
+install(TARGETS freedv DESTINATION lib)
diff --git a/libfreedv/freedv_filter.cpp b/libfreedv/freedv_filter.cpp
index 032db43c..4139a804 100644
--- a/libfreedv/freedv_filter.cpp
+++ b/libfreedv/freedv_filter.cpp
@@ -19,13 +19,14 @@
 #include <string.h>
 #include <math.h>
 #include <complex.h>
+#include <complex>
 
 #include "freedv_filter.h"
 #include "freedv_filter_coef.h"
 
 #include "fdv_arm_math.h"
 
-#define cmplx(value) (COSF(value) + SINF(value) * I)
+#define cmplx(value) (COSF(value) + SINF(value) * std::complex<float>(0.0f, 1.0f))
 
 namespace FreeDV
 {
diff --git a/libfreedv/freedv_vhf_framing.h b/libfreedv/freedv_vhf_framing.h
index 6f7dc360..3f08cbb2 100644
--- a/libfreedv/freedv_vhf_framing.h
+++ b/libfreedv/freedv_vhf_framing.h
@@ -36,6 +36,7 @@
 #ifndef _FREEDV_VHF_FRAMING_H
 #define _FREEDV_VHF_FRAMING_H
 
+#include <cstddef>
 #include "freedv_data_channel.h"
 
 /* Standard frame type */
diff --git a/libfreedv/ofdm_internal.h b/libfreedv/ofdm_internal.h
index 6c26d379..5a27e64f 100644
--- a/libfreedv/ofdm_internal.h
+++ b/libfreedv/ofdm_internal.h
@@ -41,7 +41,7 @@
 #endif
 
 #define TAU         (2.0f * M_PI)
-#define ROT45       (M_PI / 4.0f)
+#define ROT45       float(M_PI / 4.0f)
 
 #define cmplx(value) (std::complex<float>{cos(value), sin(value)})
 #define cmplxconj(value) (std::complex<float>{cos(value), -sin(value)})
```

ra1nb0w, 2019-04-14:
You are so kind. I am moved away from clang for the libfreedv error. Thank you very much! Now it works!
Do you send that patch to sdrangel or do you prefer that I send on your behalf? obviously writing that the patch is yours.

guruofquality, 2019-04-15:
Oh please send the patch to sdrangel. I will take any help I can get. Thanks!

closed 2019-04-15 by guruofquality

ra1nb0w, 2019-04-15:
ok. thanks for your attention.