pothosware/SoapySDR #220: SoapySDRDevice_setupStream compatibility stuff breaks linking when compiler decides against actually inlineing

jazzkutya, 2019-04-11:
`vrm@zivatar:~/go/src/github.com/pothosware/go-soapy-sdr/pkg/device$ go build # github.com/pothosware/go-soapy-sdr/pkg/device /tmp/go-build109665752/b001/_x003.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x004.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x005.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x006.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x007.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x008.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x009.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x010.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x011.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x012.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x013.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x014.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x015.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x016.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x017.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x018.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x019.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x020.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x021.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x022.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x023.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here /tmp/go-build109665752/b001/_x024.o: In function `SoapySDRDevice_setupStream_': /usr/local/include/SoapySDR/Device.h:310: multiple definition of `SoapySDRDevice_setupStream_' /tmp/go-build109665752/b001/_x002.o:/usr/local/include/SoapySDR/Device.h:310: first defined here collect2: error: ld returned 1 exit status 

`

can be fixed by making it static inline in Device.h:

```
|      303 static inline int SoapySDRDevice_setupStream_(SoapySDRDevice *device,
|      304     SoapySDRStream **stream,
|      305     const int direction,
|      306     const char *format,
|      307     const size_t *channels,
|      308     const size_t numChans,
|      309     const SoapySDRKwargs *args)
|-     310 {
||     311     *stream = SoapySDRDevice_setupStream(device, direction, format, channels, numChans, args);
||     312     return (*stream == NULL)?-1:0;
||     313 }

```

guruofquality, 2019-04-12:
The amount of typos that I fixed related to this one stupid function...

Thanks, its fixed!

closed 2019-04-12 by guruofquality

jazzkutya, 2019-04-12:
https://gustedt.wordpress.com/2010/11/29/myth-and-reality-about-inline-in-c99/
it seems this is actually not the proper fix, but the topis has confusion and compiler-dependent. I leave it to you to decide what to do :)

guruofquality, 2019-04-12:
It should have been static inline for a function. Otherwise you get a replication of the symbol in every file you include it in