pothosware/SoapySDR #221: Soapy refusing to compile in MinGW-w64

Kungergely, 2019-04-16:
Unfortunately tried as I can but I just can't make it compile with the minGW compiler. The problem almost certainly has something to do with C++11 support, which your code seems to pretty heavily rely on. The version of GCC/G++ it uses is 5.3.1, so it's far from being an old one, and yet I get tons of errors related to mutex and a lot of other functionality introduced in C++11. Any ideas how could I make it work?

guruofquality, 2019-04-16:
MinGW is probably not a heavily tested platform, and so maybe many of the ifdefs in either cmake or the C sources incorrectly detect windows vs unix headers and things like that, which should be easy to fix. Can you share the output of running cmake, and the output of `make VERBOSE=1`

Kungergely, 2019-04-17:
Sure! The problem seems to be present in quite a few files though.

The cmake output:
```
root@9644e80a5d71:~/rtl_433/src/SoapySDR# cmake -G "Unix Makefiles" -DCMAKE_INSTALL_PREFIX="$BUILD_HOME/build" -DCMAKE_SYSTEM_NAME=Windows -DCMAKE_SYSTEM_PROCESSOR=amd64
-- The C compiler identification is GNU 5.3.1
-- The CXX compiler identification is GNU 5.3.1
-- Check for working C compiler: /usr/bin/x86_64-w64-mingw32-gcc
-- Check for working C compiler: /usr/bin/x86_64-w64-mingw32-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/x86_64-w64-mingw32-g++
-- Check for working CXX compiler: /usr/bin/x86_64-w64-mingw32-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE)
--
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR)
-- SWIG_FOUND: FALSE -
-- Could NOT find PythonInterp (missing:  PYTHON_EXECUTABLE)
-- PYTHONINTERP_FOUND: FALSE -
-- PYTHON_EXECUTABLE: PYTHON_EXECUTABLE-NOTFOUND
-- PYTHON_INSTALL_DIR: ${prefix}/
-- Could NOT find PythonLibs (missing:  PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
-- PYTHONLIBS_FOUND: FALSE -
-- PYTHON_INCLUDE_DIRS:
-- PYTHON_LIBRARIES: PYTHON_LIBRARY-NOTFOUND
-- Performing Test SIZE_T_IS_UNSIGNED_INT
-- Performing Test SIZE_T_IS_UNSIGNED_INT - Failed
--
-- #############################################
-- ## Begin configuration for Python3 support...
-- #############################################
-- Enabling optional Python3 bindings if possible...
-- Could NOT find SWIG (missing:  SWIG_EXECUTABLE SWIG_DIR)
-- SWIG_FOUND: FALSE -
-- Could NOT find Python3Interp (missing:  PYTHON3_EXECUTABLE)
-- Could NOT find Python3InterpDbg (missing:  PYTHON3_DBG_EXECUTABLE)
-- PYTHON3INTERP_FOUND: FALSE
-- PYTHON3_EXECUTABLE: PYTHON3_EXECUTABLE-NOTFOUND
-- PYTHON3_INSTALL_DIR: ${prefix}/
-- Could NOT find Python3Libs (missing:  PYTHON3_LIBRARIES PYTHON3_INCLUDE_DIRS)
-- PYTHON3LIBS_FOUND: FALSE
-- PYTHON3_INCLUDE_DIRS: PYTHON3_INCLUDE_DIR-NOTFOUND
-- PYTHON3_LIBRARIES: PYTHON3_LIBRARY-NOTFOUND
--
-- ######################################################
-- ## SoapySDR enabled features
-- ######################################################
--
 * Library , runtime library v0.8.0-g9a942698
 * Apps , command line applications
 * Tests , library unit tests

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
--
 * Docs , doxygen documentation
 * Python , python bindings v
 * Python3 , python3 bindings
[... 624 lines cut ...]
/root/rtl_433/src/SoapySDR/lib/Device.cpp:766:10: warning: 'virtual unsigned int SoapySDR::Device::readGPIODir(const string&) const' redeclared without dllimport attribute after being referenced with dll linkage
 unsigned SoapySDR::Device::readGPIODir(const std::string &) const
          ^
/root/rtl_433/src/SoapySDR/lib/Device.cpp:775:6: warning: 'virtual void SoapySDR::Device::writeI2C(int, const string&)' redeclared without dllimport attribute: previous dllimport ignored [-Wattributes]
 void SoapySDR::Device::writeI2C(const int, const std::string &)
      ^
/root/rtl_433/src/SoapySDR/lib/Device.cpp:780:13: warning: 'virtual std::__cxx11::string SoapySDR::Device::readI2C(int, size_t)' redeclared without dllimport attribute: previous dllimport ignored [-Wattributes]
 std::string SoapySDR::Device::readI2C(const int, const size_t)
             ^
/root/rtl_433/src/SoapySDR/lib/Device.cpp:788:10: warning: 'virtual unsigned int SoapySDR::Device::transactSPI(int, unsigned int, size_t)' redeclared without dllimport attribute: previous dllimport ignored [-Wattributes]
 unsigned SoapySDR::Device::transactSPI(const int, const unsigned, const size_t)
          ^
/root/rtl_433/src/SoapySDR/lib/Device.cpp:796:26: warning: 'virtual std::vector<std::__cxx11::basic_string<char> > SoapySDR::Device::listUARTs() const' redeclared without dllimport attribute: previous dllimport ignored [-Wattributes]
 std::vector<std::string> SoapySDR::Device::listUARTs(void) const
                          ^
/root/rtl_433/src/SoapySDR/lib/Device.cpp:801:6: warning: 'virtual void SoapySDR::Device::writeUART(const string&, const string&)' redeclared without dllimport attribute: previous dllimport ignored [-Wattributes]
 void SoapySDR::Device::writeUART(const std::string &, const std::string &)
      ^
/root/rtl_433/src/SoapySDR/lib/Device.cpp:806:13: warning: 'virtual std::__cxx11::string SoapySDR::Device::readUART(const string&, long int) const' redeclared without dllimport attribute: previous dllimport ignored [-Wattributes]
 std::string SoapySDR::Device::readUART(const std::string &, const long) const
             ^
make[1]: *** [lib/CMakeFiles/SoapySDR.dir/all] Error 2
make: *** [all] Error 2

```

Kungergely, 2019-04-17:
Moreover I've tried to use all sorts of gimmicks for telling g++ to enable C++11 support (via passing -std=c++0x or -std=c++11 to g++), but to no avai. The errors have always remained the same.

guruofquality, 2019-04-17:
`'recursive_mutex' in namespace 'std' does not name a type`

From the looks of the errors, its flat out just missing important C++ features in the development headers.

I saw this thread, and its somewhat old, but very reminiscent of the problem: https://stackoverflow.com/questions/14191566/c-mutex-in-namespace-std-does-not-name-a-type

Perhaps there is a checkbox for thread support package when installing mingw

righthalfplane, 2019-12-09:
It looks like the only know work around is to use -

https://github.com/meganz/mingw-std-threads

and that is not so hot.

Kungergely, 2019-12-09:
> 
> and that is not so hot.
Actually including this seems to be literally painful in the Travis/Docker build automation scheme I've set up. I'll see if I can make it work using some well-placed loops, sed commands and stuff. Thanks for the find :)

Kungergely, 2019-12-09:
> From the looks of the errors, its flat out just missing important C++ features in the development headers.

Yeah.
> 
> Perhaps there is a checkbox for thread support package when installing mingw

Unfortunately that would be of little help when I'm using Linux to make the SoapySDR builds for Windows.

righthalfplane, 2019-12-10:
Using MSYS2 and mingw64.exe on windows 7, I was able to build SoapySDR and a working version of SoapySDRUtil.exe, which found the 3 SDRs that I had connected. With the cmake option -

cmake -G"MSYS Makefiles" ../

from the directory SoapySDR/build. I tied to install the support for NetSDR, but that failed. Maybe the driver for the device you want will work.

righthalfplane, 2019-12-14:
I got HackRF One to install under MSYS2 and played some stations with it.

Kungergely, 2019-12-14:
> Using MSYS2 and mingw64.exe on windows 7

I need to build this on Linux, not Windows. Automating anything like this is a nightmare on Windows, besides, I really don't feel like redoing the whole build process for Windows.

righthalfplane, 2019-12-24:
Are you trying to compile programs for windows on a linux machine ?

Kungergely, 2019-12-24:
I already am compiling rtl_433 for Windows, x86_64, 32-bit and 64-bit ARM Linux on Linux machines, yeah. Currently the Linux builds have both SoapySDR and RTL-SDR compiled in, but the Windows build has only the latter.

closed 2022-11-24 by ncorgan