pothosware/SoapySDR #222: module 'SoapySDR' has no attribute 'Device'

JloveL, 2019-04-18:
(Conda3) atticus@MIR:~$ python -m qspectrumanalyzer
soapy_power params:
{'bandwidth': 0.0,
 'bin_size': 10.0,
 'crop': 0.0,
 'device': '',
 'gain': 37,
 'hops': 0,
 'interval': 10.0,
 'ppm': 0,
 'sample_rate': 2560000.0,
 'single_shot': False,
 'start_freq': 87.0,
 'stop_freq': 108.0}

Traceback (most recent call last):
  File "/home/atticus/Conda3/bin/soapy_power", line 10, in <module>
    sys.exit(main())
  File "/home/atticus/Conda3/lib/python3.6/site-packages/soapypower/__main__.py", line 323, in main
    output_format=args.format
  File "/home/atticus/Conda3/lib/python3.6/site-packages/soapypower/power.py", line 38, in __init__
    force_sample_rate=force_sample_rate, force_bandwidth=force_bandwidth
  File "/home/atticus/Conda3/lib/python3.6/site-packages/simplesoapy.py", line 78, in __init__
    self.device = SoapySDR.Device(soapy_args)
AttributeError: module 'SoapySDR' has no attribute 'Device'

JloveL, 2019-04-18:
anybody helps me？

guruofquality, 2019-04-19:
Never seen this before:

* How did you install SoapySDR?
* Where are the python files for SoapySDR installed?
* Which files are present in this directory?
* What are the contents of these files?

For example:
```
ls /usr/local/lib/python3.6/dist-packages
SoapySDR.py  _SoapySDR.so

cat /usr/local/lib/python3.6/dist-packages/SoapySDR.py | grep "class Device"
class Device(_object):
class Device(Device):
```

JloveL, 2019-04-19:
I followed [this](https://travis-ci.org/pothosware/SoapySDR/jobs/519018586) to install Soapy ,and the test passed :

> atticus@MIR:~/Conda3/lib/python3.6/site-packages/SoapySDR/build$  make test
> Running tests...
> Test project /home/atticus/Conda3/lib/python3.6/site-packages/SoapySDR/build
> Start 1: TestTimeConversion
> 1/4 Test #1: TestTimeConversion ...............   Passed    0.00 sec
> Start 2: TestFormatParser
> 2/4 Test #2: TestFormatParser .................   Passed    0.00 sec
> Start 3: TestKwargsMarkup
> 3/4 Test #3: TestKwargsMarkup .................   Passed    0.00 sec
> Start 4: TestConvertTypes
> 4/4 Test #4: TestConvertTypes .................   Passed    0.00 sec
> 
> 100% tests passed, 0 tests failed out of 4
> 
> Total Test time (real) =   0.01 sec
> 

SoapySDR installed here:

> atticus@MIR:~/Conda3/lib/python3.6/site-packages/SoapySDR$ ls
> apps           cmake                     ExampleDriver    python3
> appveyor.yml   CMakeLists.txt            include          README.md
> build          cmake_uninstall.cmake.in  lib              tests
> buildkite      debian                    LICENSE_1_0.txt
> Changelog.txt  docs                      python
> 

thanks for your reply

guruofquality, 2019-04-19:
> SoapySDR installed here
> atticus@MIR:~/Conda3/lib/python3.6/site-packages/SoapySDR

But this is just the source directory and it looks like build files in a weird location. Can you follow the build guide, make install will put files where they are supposed ot be: https://github.com/pothosware/SoapySDR/wiki/BuildGuide#unix-instructions

JloveL, 2019-04-20:
thanks for reply.
I folloewed the guide above, when i type:
> SoapySDRUtil --info
I got this:

> atticus@MIR:~/SoapySDR/build$ SoapySDRUtil --info
> ######################################################
> ##     Soapy SDR -- the SDR abstraction library     ##
> ######################################################
> 
> Lib Version: v0.8.0-g9a942698
> API Version: v0.8.0
> ABI Version: v0.8
> Install root: /usr/local
> Search path:  /usr/local/lib/SoapySDR/modules0.8 (missing)
> No modules found!
> Available factories... No factories found!
> Available converters...
>  -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
>  -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
>  -  CS32 -> [CS32]
>  -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
>  -  CU16 -> [CF32, CS16, CS8]
>  -   CU8 -> [CF32, CS16, CS8]
>  -   F32 -> [F32, S16, S8, U16, U8]
>  -   S16 -> [F32, S16, S8, U16, U8]
>  -   S32 -> [S32]
>  -    S8 -> [F32, S16, S8, U16, U8]
>  -   U16 -> [F32, S16, S8]
>  -    U8 -> [F32, S16, S8]
Search path:  /usr/local/lib/SoapySDR/modules0.8 (missing)

thanks for reply.

guruofquality, 2019-04-20:
the basic installation looks good, but it does not change the issue that `~/Conda3/lib/python3.6/site-packages/SoapySDR` is NOT a real python module, and for some reason it seems to be in your python's search path for modules. You have to fix that.

closed 2019-04-24 by guruofquality