pothosware/SoapySDR #223: SoapySDRDevice_readStream returned -6: TIME_ERROR

BilalMaqsood88, 2019-04-23:
Hi SoapySDR Team,

I am using soapySDR with srsLTE (https://github.com/srsLTE/srsLTE) to access the LimeSDR usb. While running the srsUE, I recieve this error, can you please let me know the cause and it will be great if you suggest what could be done.  
 
SoapySDRDevice_readStream returned -6: TIME_ERROR

Thanks and Regards,
Bilal

guruofquality, 2019-04-23:
My guess is that the application is passing a time to activateStream() that is less than the current time on the device.

BilalMaqsood88, 2019-04-23:
Thanks for the reply. Any idea how to fix this ?

guruofquality, 2019-04-24:
Put some debug prints into srsLTE, and try to figure out why they are requesting late timestamps. If srsLTE is fine, but it appears to be a bug in LimeSuite, try to make a test case to show the failure so the folks at Lime Micro can fix it - https://github.com/myriadrf/LimeSuite/issues

closed 2019-04-24 by guruofquality