pothosware/SoapySDR #225: python binding docs in wiki encourages bad practice

jazzkutya, 2019-05-04:
Hi,

in the python docs:

> Some objects contain references to “external” resources such as open files or windows. It is understood that these resources are freed when the object is garbage-collected, but since garbage collection is not guaranteed to happen, such objects also provide an explicit way to release the external resource, usually a close() method. Programs are strongly recommended to explicitly close such objects.

this implies the following in the [wiki](https://github.com/pothosware/SoapySDR/wiki/PythonSupport) is bad practice (regarding uname()):

>The Device::make() and Device::unmake() factory functions are not used in Python. Rather, the Device constructor directly calls into the make() factory, and the Device destructor directly calls into unmake() for cleanup.

now I see from the python code generated by swig that we can use the class method Device.unmae() or the function Device_unmake(), but it should be explicitly mentioned in the wiki and I don't know how official they are. Or which one is the official way to release the external resources of a Device.

This may be just a documentation issue on the wiki requiring no change of code.

guruofquality, 2019-05-04:
The python object for device in this case is reference counted. Removing all references to the device will cause it to be cleaned up. This happens whenever every copy of the device goes out of scope or is explicitly cleared, by setting it to none or some other variable. For example:

```
d = SoapySDR.Device() #make called here

#stuff happens here

d = None #unmake called here
```

jazzkutya, 2019-05-04:
I know what, but it's too easy to have lingering references to it from a cyclic structure. And the python docs itself says it's best to use the close() method explicitly, which implies SoapySDR binding should have a documented official close() or unmake() function or method.

jazzkutya, 2019-05-04:
i see i forgot to link the source of the quote from the wiki, will edit it in soon

jazzkutya, 2019-05-04:
I've linked the python doc in this issue.
more interesting quotes from the same python doc:

>Objects are never explicitly destroyed; however, when they become unreachable they **may** be garbage-collected. An implementation is allowed to postpone garbage collection or omit it altogether — it is a matter of implementation quality how garbage collection is implemented, as long as no objects are collected that are still reachable.

Now it is true that CPython is documented to use reference counting but detection of cyclically linked garbage is optional and delayed. While I guess it is true that the SoapySDR binding is specific to CPython I don't think that is enough to go against the general stance of the language in this matter.

guruofquality, 2019-05-04:
something like this:
```
diff --git a/python/SoapySDR.i b/python/SoapySDR.i                                                                                                                                                                                           
index e6cca20..bbbf2ed 100644                                                                                                                                                                                                                
--- a/python/SoapySDR.i                                                                                                                                                                                                                      
+++ b/python/SoapySDR.i                                                                                                                                                                                                                      
@@ -313,9 +313,14 @@ def extractBuffPointer(buff):                                                                                                                                                                                           
                                                                                                                                                                                                                                             
     %insert("python")                                                                                                                                                                                                                       
     %{                                                                                                                                                                                                                                      
+        def close(self):                                                                                                                                                                                                                    
+            Device.unmake(self)                                                                                                                                                                                                             
+            setattr(self, '__closed__', True)                                                                                                                                                                                               
+                                                                                                                                                                                                                                            
         #call unmake from custom deleter                                                                                                                                                                                                    
         def __del__(self):                                                                                                                                                                                                                  
-            Device.unmake(self)                                                                                                                                                                                                             
+            try: getattr(self, '__closed__')                                                                                                                                                                                                
+            except AttributeError: Device.unmake(self)
 
         def __str__(self):
             return "%s:%s"%(self.getDriverKey(), self.getHardwareKey())
```

jazzkutya, 2019-05-04:
thanks, looks good and I made a test. Properly guards against double unmake()

closed 2019-05-25 by jazzkutya