pothosware/SoapySDR #227: Windows file separators incorrect in several places

MarcKaufman, 2019-06-03:
There are several places in Modules.cpp where the incorrect separator is used. Here are my patches:

SoapySDR::listSearchPaths
    searchPaths.push_back(SoapySDR::getRootPath() + 
#ifdef _MSC_VER
		"\\lib\\SoapySDR\\modules"
#else
		"/lib/SoapySDR/modules"
#endif
		+ SoapySDR::getABIVersion());

SoapySDR::listModules
   return searchModulePath(path + 
#ifdef _MSC_VER
		"\\"
#else
		"/"
#endif
	); //requires trailing slash

guruofquality, 2019-06-03:
Most command lines don't allow "/" because it conflicts with the use of "/" for a switch delimiter. But the Windows and DOS kernels are happy with either "\" or "/". So something isnt broken, this is just aesthetic right?

MarcKaufman, 2019-06-03:
Nothing to do with the command line. getRootPath and listModules are used for finding the dlls to load. If you don't do the change you get "/" in Windows paths, which doesn't work. I was chasing a bug where SDRPlaySupport wouldn't load in CubicSDR. Cubic was giving Soapy a path ending in "\\\\", which then had a "/" appended, making file search unworkable. I'll try taking out the Soapy changes one at a time now that I have the file search running correctly.

MarcKaufman, 2019-06-03:
I just checked with the other delimiters. Works fine. Sorry for the false bug report.

closed 2019-06-03 by MarcKaufman