pothosware/SoapySDR #228: SoapySDR build failure on Python 3.8

mdomsch, 2019-06-03:


guruofquality, 2019-06-03:
* reference https://copr-be.cloud.fedoraproject.org/results/@python/python3.8/fedora-rawhide-x86_64/00917251-SoapySDR/build.log.gz

From the log it looks like SoapySDR is just not getting linked to python. I think this is a fedora bug in python3-config, for example, python3 on ubuntu 18.04 reports the link libraries including libpython:
```
python3-config --ldflags
-L/usr/lib/python3.6/config-3.6m-x86_64-linux-gnu -L/usr/lib -lpython3.6m -lpthread -ldl  -lutil -lm  -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
```

But on Fedora its only showing:
```
-L/usr/lib64  -lcrypt -lpthread -ldl  -lutil -lm -lm
```

Your other option is to pass an acceptable `-DPYTHON3_LIBRARY=path_to_the_lib` to cmake

hroncok, 2019-06-03:
Python 3.8 no longer builds extension modules against libpython. See https://docs.python.org/dev/whatsnew/3.8.html#debug-build-uses-the-same-abi-as-release-build

guruofquality, 2019-06-03:
> On the other hand, pkg-config python3.8 --libs no longer contains -lpython3.8. C extensions must not be linked to libpython

Weird, ok. So I think the issue is that I need to remove `-Wl,--no-undefined` to stop all of those linker errors. And then presumably everything will work at runtime because of python3. (The no-undefined was useful with previous python versions to catch stuff like this) Assuming that is the problem anyway...

Is anyone in a state where they can test the removal of this flag on python3.8?
* https://github.com/pothosware/SoapySDR/blob/master/lib/CMakeLists.txt#L78

hroncok, 2019-06-03:
We have an older version in Fedora, but sedding it out of `SoapySDRConfig.cmake` works:

    sed -i 's/ -Wl,--no-undefined//g' cmake/Modules/SoapySDRConfig.cmake

mdomsch, 2019-06-03:
I've built 0.7.1 into Fedora rawhide (will be 31) with the above patch that removes -Wl,--no-undefined in the spec file, before the build commences.  This built properly.

guruofquality, 2019-06-03:
Follow up question, after it built did the module import? This simple command line should do it, should echo 0

```
python3 -c "import SoapySDR" && echo $?
```

hroncok, 2019-06-03:
yes (with SoapySDR-0.6.1-4.20180806gite694813.fc31 Fedora package)

hroncok, 2019-06-03:
same with 0.7.1

guruofquality, 2019-06-04:
I removed this flag on the maint branch which will be tagged as 0.7.2 and on the master for future major releases.

mdomsch, 2019-06-04:
Thank you for the speedy investigation and solution.

closed 2019-06-04 by mdomsch