pothosware/SoapySDR #229: Segmentation fault (core dumped) on SoapySDRDevice_activateStream

David-mtl, 2019-06-18:
Hi, 
I'm using the C example from here: https://github.com/pothosware/SoapySDR/wiki/C_API_Example.
I'm getting a segmentation fault on the call to SoapySDRDevice_activateStream. 
I run the code on ubuntu 18.04. 
I use the LimeSDR Mini.
The output in the console is:

```
gcc -std=c99 example.c -lSoapySDR -lm && ./a.out
Found device #0: addr=24607:1027, driver=lime, label=LimeSDR Mini [USB 3.0] 1D4C36DC679185, media=USB 3.0, module=FT601, name=LimeSDR Mini, serial=1D4C36DC679185, 
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D4C36DC679185'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M calibration values caching Disable
Rx antennas: NONE, LNAH, LNAL_NC, LNAW, Auto, 
Rx gains: TIA, LNA, PGA, 
Rx freq ranges: [0 Hz -> 3.8e+09 Hz], 
[INFO] Selected RX path: LNAW
Segmentation fault (core dumped)

```
Any idea on how to fix this?

guruofquality, 2019-06-18:
Other than modifying the example for driver = lime, it worked out of the box for me with latest master branch of LimeSuite, SoapySDR, ubuntu 18.04 on limesdr mini. Double check what you installed. For example: https://github.com/pothosware/SoapySDR/wiki/ConfigGuide#avoid-simultaneous-installs And when in doubt, debug with backtraces, and printfs.

David-mtl, 2019-06-18:
You are right, I removed all packages for LimeSuite and SoapySDR then rebuild both project from source and now it's working properly. Thanks.

closed 2019-06-18 by David-mtl