pothosware/SoapySDR #230: C_API Compile Error: too few arguments to function 'SoapySDRDevice_setupStream'

PS-FITT, 2019-07-01:
Hi,

i´m new to this stuff and so I tried to compile the C_API_Example on a raspi3 with gcc and get following Error:
too few arguments to function 'SoapySDRDevice_setupStream'.......
SoapySDRStream *rxStream = SoapySDRDevice_setupStream(sdr, SOAPY_SDR_RX, SOAPY_SDR_CF32, NULL, 0, NULL);

At this point I know there is an argument in the source missing and verified it in /usr/local/include/SoapySDR/Device.h:294:19. There are seven "Vars" defined and six vars are given.OK that I understand but I don´t know where to find what is missing,  I think "SoapySDRStream **stream," the second argument is missing but I´m not sure and also don´t know how to test.

What I want: I need a solution to set on rasperry Startup the Frequency the RSPA1 listen to and give the stream to another programm which make some actions if there comes some information on the stream. Ok it would be nice if the stream would be on the Line-IN but it´s on USB. 

RSP_TCP Server is near the solution and so I decided to test this c_api things and get the error. Someone can help?

guruofquality, 2019-07-01:
Maybe you have found the first compiler that doesnt like this macro. You can actually delete this entire section and you should get the intended normal behaviour. It just exists to offer some code backwards compatibility by overloading on the number of args.

Can you tell me a few things about your system? Its raspi -- so which distro or OS, whats the version of gcc? `gcc --version`

TRY TO DELETE THIS in Device.h:
```
///@cond INTERNAL
//! Old setupStream invoked by compatibility macro
static inline int SoapySDRDevice_setupStream_(SoapySDRDevice *device,
    SoapySDRStream **stream,
    const int direction,
    const char *format,
    const size_t *channels,
    const size_t numChans,
    const SoapySDRKwargs *args)
{
    *stream = SoapySDRDevice_setupStream(device, direction, format, channels, numChans, args);
    return (*stream == NULL)?-1:0;
}

#define SoapySDR_getMacro(_1,_2,_3,_4,_5,_6,_7,NAME,...) NAME
#define SoapySDRDevice_setupStream(...) SoapySDR_getMacro(__VA_ARGS__, SoapySDRDevice_setupStream_, SoapySDRDevice_setupStream)(__VA_ARGS__)
///@endcond
```

PS-FITT, 2019-07-01:
hi and thx for reply.
I will/can test it tomorrow with the delete in device.h (OK in the line is descriped "Old setup stream" ;-) )

OK my system infos 👍

gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516
on a Raspberry Pi 3 Model B Plus with
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian

so i will reply again after tests

PS-FITT, 2019-07-02:
Hi again and GoodMorning,

ok sorry it was my mistake in case of using the wrong Device.h.

I installed the Raspi 0.5 image from sdrplay Website with the SDREasyPlay things and on this there is a "Device.h" also in /usr/local/include/SoapySDR/ which is different to your Device.h from this git repo.

So I copied the Device.h of this git repo into the /usr/local/include/SoapySDR Path and the compilation works without deleting the lines you have mentioned.
[Device.h.orig.zip](https://github.com/pothosware/SoapySDR/files/3348762/Device.h.orig.zip)

I attached the Device.h which won´t work if you want to have a look at it.
 Sorry again didn´t mention the include-Path and thx.

closed 2019-08-01 by guruofquality