pothosware/SoapySDR #231: sdrplay and kernel errors.

starvald, 2019-07-05:
I've had this problem for a while but only with the soapysdr enabled version of rtl_airband, gqrx and cubicsdr are unaffected.
i will see kernel taint errors until rtl_airband will eventually die.

used to see this problem with the drivers but sdrplay fixed over a year ago so it's interesting that rtl_airband seems to still be triggering them.

[dmesg.txt](https://github.com/pothosware/SoapySDR/files/3363559/dmesg.txt)

starvald, 2019-07-07:
Just noticed soapy was on 0.7, updated to 0.8.
will see how it goes.

guruofquality, 2019-07-07:
> Just noticed soapy was on 0.7, updated to 0.8.
will see how it goes.

Not likely. SoapySDR knows absolutely nothing about usb or kernel modules.

This is going to either be an issue with the underlying driver like librtlsdr or maybe some underlying kernel driver for usb.

Its always possible its how SoapyRTL uses the librtlsdr driver, like in terms of parameters passed to librtlsdr -- so there could potentially be an issue there.

starvald, 2019-07-20:
Yeah still get kernel taint and a segfault, opening a sdrplay support ticket.

starvald, 2019-07-21:
sdrplay support replied and they don't believe it's driver related and suggest it's an rtl_airband problem.

they did suggest "enable debug by adding a call to mir_sdr_DebugEnable(1); before the mir_sdr_StreamInit call in the SoapySDRPlay library" to help debug it.

guruofquality, 2019-08-25:
Not sure what to do, open an issue with https://github.com/pothosware/SoapySDRPlay/issues if needed

closed 2019-08-25 by guruofquality