pothosware/SoapySDR #234: Non-obvious limitations for SoapySDRUtil --rate --args

metssigadus, 2019-08-06:
Summary: SoapySDRUtil --rate command will not permit remote port and serial number prsesented as concurrent arguments. No error msg, no crashes, but the functionality drops - the rate test never starts.

Test setup: Ubuntu 18.04 netinstall amd64 with a minimalistic setup, Soapy prepackaged, except obviously the SoapySDRPlay which has been compiled locally from git sources.

SoapySDRUtil --find
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Found device 0
  driver = remote
  label = SDRplay Dev0 RSP1A 180700DC93
  remote = tcp://10.10.1.142:55132
  remote:driver = sdrplay
  serial = 180700DC93

Found device 1
  driver = sdrplay
  label = SDRplay Dev0 RSP1A 180700DC93
  serial = 180700DC93

Now launching ratetest with variations in arguments:

This works:  **SoapySDRUtil --rate=1e6 --direction=RX** 
This works:  **SoapySDRUtil --rate=1e6 --direction=RX --args="driver=remote"**
This works:  **SoapySDRUtil --rate=1e6 --direction=RX --args="driver=remote,remote=tcp://10.10.1.142:55132"**
This does not work: **SoapySDRUtil --rate=1e6 --direction=RX --args="driver=remote,remote=tcp://10.10.1.142:55132,serial=180700DC93"**
The stream never starts.
For comparison, the command
**SoapySDRUtil --probe="driver=remote, remote=tcp://10.10.1.142:55132, serial=180700DC93"**
works nice with the same set or arguments.

Even this does not work: **SoapySDRUtil --rate=1e6 --direction=RX --args="serial=180700DC93"**
And this does not work: **SoapySDRUtil --rate=1e6 --direction=RX --args="remote=tcp://10.10.1.142:55132, serial=180700DC93"**
But this does work: **SoapySDRUtil --rate=1e6 --direction=RX --args="driver=remote, serial=180700DC93"**

I do not see the orderly logic behind this behaviour. Can plz somebody hint me one.

This test was attempted due to an advice to launch two SoapyServers on the radio host, on various ports, to overcome the known limitations of the SoapySDRPlay module (risk of crash if accessed by two concurrent processes).

guruofquality, 2019-08-07:
For what its worth the arguments look fine and its working properly here for "SDRplay Dev0 RSPduo" on an ubuntu 18.04 server and 18.04 client machine. Is there any particular error message printed when "it does not work" like device not found?

If you are looking to dig deeper, this is the code that asks the mirics API for devices and reports serial numbers: https://github.com/pothosware/SoapySDRPlay/blob/master/Settings.cpp#L33

And this is the code that open the device given its serial number: https://github.com/pothosware/SoapySDRPlay/blob/master/Settings.cpp#L33

As long as there is a device argument that narrows down the criteria to a particular or singular device, you should end up in the same place in the code.

Also this is a tracker for SDRPlay issues: https://github.com/pothosware/SoapySDRPlay/issues

metssigadus, 2019-08-07:
OK. Trying to rephrase. You are saying a) all these argument rows are  legal and b) they are actually effective at another setup elsewhere.
> its working properly here for "SDRplay Dev0 RSPduo" on an ubuntu 18.04 server and 18.04 client machine.
That's the most valuable piece of information. I will try to variate other conditions to the extent I can.

> Is there any particular error message printed when "it does not work" like device not found?
Not, there isn't. At that moment when normally the "propeller" starts "revolving" and Msps/Mbps values do appear, nothing happens. As of the program is waiting for something that never arrives. And so for at least 30 secs, while normally the stream starts within 2-3 secs. No error messages, no 0 Msps messages, no printout at all. Till the control-C normally ends the program (no error messages again). 

> Also this is a tracker for SDRPlay issues: https://github.com/pothosware/SoapySDRPlay/issues
Yes, I have seen all the rich quantity pothosware modules. But, due to my almost nonexisting coding skills, unable to locate the proper package.

metssigadus, 2019-08-07:
Plz take your time, this is long but definitive.

Now when I have obtained the forehand knowledge about my argument lines being correct, and the absolute certainty they work elsewhere, I was able to variate the other parameters. One of the first ideas was timing. By Linux shell ARROW UP function it is possible to repeat the command rapidly, right after the  previous CTL-C returned to the shell prompt (with a delay around 0.5-0.8 sec). Alternatively, it is possible to keep some longer pause between the commands (5 sec or e.g. 15 sec).

The very first conclusion when fast re-launching the ratetest command (tens of times), was, that the stability is low. The average probablity that the command will succeed (and not stop without ever meeting the stream) is 30% to 90%, but never 100%.

I also tried to launch ratetest commands under strace and noticed almost instantly that the success rate is considerably higher - with tens of relaunches, I only was able to meet one failure to stream. Thus I got the idea - the problem has to do with timing and likely the **relaunch timing does matter**.
 
The machine is (during all the test of this issue) a Lenovo M-series Desktop, with an i5 processor and 6GB of memory, OS is a minimal stock Ubuntu 18.04 with SoapySDRPlay module compiled locally. I really believe, it is a machine powerful enough for the purpose (smpboot: Total of 4 processors activated (26340.00 BogoMIPS)). I occasionally looked at the load, was able to spot 0.26 while average at 0.05.

Then I started (manually) to repetitively re-launch the ratetest commands, using all the 7 examples above. The result is unexpected - all these commands work ... sometimes .... with a certain probability. None of these works perfectly and at any time. A simple command I was sure it works -  **SoapySDRUtil --rate=1e6 --direction=RX** -  actually is working with 80-90% reliability. A complex multi-argument command I was sure it does not work - **SoapySDRUtil --rate=1e6 --direction=RX --args="driver=remote,remote=tcp://10.10.1.142:55132,serial=180700DC93"** -, in fact works, but with a reliability of 30-50%.

Then I started to variate the pauses between the re-launch of the same command - they have a strong effect to the outcome. By keeping the pause very short (0.5s) I was able to see 3-4 consecutive successful relaunches of the complex (multi-argument) command. It seems, like the worst timing region is between 0.5 and 4 seconds. The failure rate is highest there. There are two local optimums for a pause duration: less than 0,5 sec (for lengthy argument lines) and over 15s for the rest. Aiming these optimums, the successful re-launch rate will reach it's maximum, approaching the levels indicated above (but depending on the arguments length). There was NO ERROR at the CLI or in the syslog. There was one case when CLI started to print capital S's (mentioned in an earlier issue).

**_The summary: FIRST, I apologise for saying it in black/white that certain commands do not work. In fact, their success is probabilistic, not deterministic. It seems the timing strongly correlates with the outcome (well, vice versa). And SECOND - whatever are the problems or in which layer, they seriously interfere with a successful usage._** 

I have a logfile (3.4M packed) of my tests. Actually, this test should be conducted at an automated testbed, with thousands of re-launches and with automatic measurements of the outcome. Unsuccessfully I have no such tools at my home.

Addendum: the output of a) one failed, b) one succeeded command (success appeared only on the third re-launch) and c) an occasion with capital S's printed:

air@air3:~$ SoapySDRUtil --rate=1e6 --direction=RX --args="driver=remote,remote=tcp://10.10.1.142:55132,serial=180700DC93"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

[INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CS16, scaleFactor=32767, mtu=1500, window=44040192)
[INFO] Client side stream bound to 10.10.1.142:40459
[INFO] Client side status bound to 10.10.1.142:54531
[INFO] Using format CS16.
[INFO] Server side stream bound to [::ffff:10.10.1.142]:58408
[INFO] Server side stream connected to [::ffff:10.10.1.142]:40459
[INFO] Server side status connected to [::ffff:10.10.1.142]:54531
[INFO] Configured sender endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=43008 KiB
[INFO] Client side stream connected to 10.10.1.142:58408
[INFO] Configured receiver endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=43008 KiB
Stream format: CS16
Num channels: 1
Element size: 4 bytes
Begin RX rate test at 1 Msps
Starting stream loop, press Ctrl+C to exit...
**[---- Here is the place when nothing will happen forever ----]**
^Cair@air3:~$ SoapySDRUtil --rate=1e6 --direction=RX --args="driver=remote,remote=tcp://10.10.1.142:55132,serial=180700DC93"
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

[INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CS16, scaleFactor=32767, mtu=1500, window=44040192)
[INFO] Client side stream bound to 10.10.1.142:47180
[INFO] Client side status bound to 10.10.1.142:56676
[INFO] Using format CS16.
[INFO] Server side stream bound to [::ffff:10.10.1.142]:51990
[INFO] Server side stream connected to [::ffff:10.10.1.142]:47180
[... cut ...]
[INFO] Configured sender endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=43008 KiB
[INFO] Client side stream connected to 10.10.1.142:53250
[INFO] Configured receiver endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=43008 KiB
Stream format: CS16
Num channels: 1
Element size: 4 bytes
Begin RX rate test at 1 Msps
Starting stream loop, press Ctrl+C to exit...
SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSS^Cair@air3:~$

closed 2019-08-25 by guruofquality