pothosware/SoapySDR #238: Can SoapySDR handle10G streaming?

sirjansel0t, 2019-09-17:
Hello,

I am trying to determine if SoapySDR can handle 10G streaming I/O.  I am trying to stream samples via 10G ethernet to be able to get rates above 60 MSPS.  Can I use SoapySDR to do that?  Does SoapySDR support Zero Copy buffers and is that the way to go?  

Thank you.

guruofquality, 2019-09-18:
SoapySDR is just an abstraction for drivers, so its only overhead is really just a function call. There is a function call "readStream" that fills in a user provided buffer. And there is one called "acquireReadBuffer" that provided a driver buffer to the user. If you can write a driver for either one of those functions and a matching application, then you are set.

FWIW I have used the API to get well over 10G rates on network devices, on the same machine, but this was in aggregate over multiple channels, not one stream in this case -- if that helps.

sirjansel0t, 2019-09-18:
Ok got it.  Thank you.

closed 2019-09-18 by sirjansel0t