pothosware/SoapySDR #239: Installing in virtualenv not trivial

gasparka, 2019-09-20:
Reproducer:
```bash
git clone https://github.com/pothosware/SoapySDR
cd SoapySDR
virtualenv ./venv
source venv/bin/activate
mkdir build
cd build
cmake ..
```

Cmake output, with error:
```bash
-- Build type not specified: defaulting to release.
-- 
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- SWIG_FOUND: TRUE - 4.0.1
-- PYTHONINTERP_FOUND: TRUE - 3.7.4
-- PYTHON_EXECUTABLE: /home/gaspar/git/SoapySDR/venv/bin/python
Traceback (most recent call last):
  File "/home/gaspar/git/SoapySDR/python/get_python_lib.py", line 10, in <module>
    paths = [p for p in site.getsitepackages() if p.startswith(prefix)]
AttributeError: module 'site' has no attribute 'getsitepackages'
-- PYTHON_INSTALL_DIR: ${prefix}/
-- PYTHONLIBS_FOUND: TRUE - 3.7.4
-- PYTHON_INCLUDE_DIRS: /home/gaspar/git/SoapySDR/venv/include/python3.7m;/home/gaspar/git/SoapySDR/venv/include/python3.7m
-- PYTHON_LIBRARIES: -lpython3.7m -lcrypt -lpthread -ldl -lutil -lm -Xlinker -export-dynamic
-- CMAKE_SWIG_FLAGS=-c++;-threads;-I/home/gaspar/git/SoapySDR/include
CMake Warning (dev) at /usr/share/cmake-3.15/Modules/UseSWIG.cmake:607 (message):
  Policy CMP0078 is not set: UseSWIG generates standard target names.  Run
  "cmake --help-policy CMP0078" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

Call Stack (most recent call first):
  python/CMakeLists.txt:155 (SWIG_ADD_LIBRARY)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.15/Modules/UseSWIG.cmake:460 (message):
  Policy CMP0086 is not set: UseSWIG honors SWIG_MODULE_NAME via -module
  flag.  Run "cmake --help-policy CMP0086" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/share/cmake-3.15/Modules/UseSWIG.cmake:702 (SWIG_ADD_SOURCE_TO_MODULE)
  python/CMakeLists.txt:155 (SWIG_ADD_LIBRARY)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at python/CMakeLists.txt:164 (install):
  install TARGETS given no LIBRARY DESTINATION for module target "_SoapySDR".

CMake Error at python/CMakeLists.txt:169 (install):
  install FILES given no DESTINATION!

-- 
-- #############################################
-- ## Begin configuration for Python3 support...
-- #############################################
-- Enabling optional Python3 bindings if possible...
-- SWIG_FOUND: TRUE - 4.0.1
-- Could NOT find Python3InterpDbg (missing: PYTHON3_DBG_EXECUTABLE) 
-- PYTHON3INTERP_FOUND: TRUE
-- PYTHON3_EXECUTABLE: /home/gaspar/git/SoapySDR/venv/bin/python3
Traceback (most recent call last):
  File "/home/gaspar/git/SoapySDR/python3/../python/get_python_lib.py", line 10, in <module>
    paths = [p for p in site.getsitepackages() if p.startswith(prefix)]
AttributeError: module 'site' has no attribute 'getsitepackages'
-- PYTHON3_INSTALL_DIR: ${prefix}/
-- PYTHON3LIBS_FOUND: TRUE
-- PYTHON3_INCLUDE_DIRS: /usr/include/python3.7m;/usr/include/python3.7m
-- PYTHON3_LIBRARIES: -L/usr/lib -lpython3.7m -lcrypt -lpthread -ldl  -lutil -lm
-- 
-- ######################################################
-- ## SoapySDR enabled features
-- ######################################################
-- 
 * Library, runtime library v0.8.0-gcafa4ba5
 * Apps, command line applications
 * Tests, library unit tests
 * Docs, doxygen documentation
 * Python, python bindings v3.7.4

-- ######################################################
-- ## SoapySDR disabled features
-- ######################################################
-- 
 * Python3, python3 bindings

-- SoapySDR version: v0.8.0-gcafa4ba5
-- ABI/so version: v0.8
-- Install prefix: /usr/local
-- Configuring incomplete, errors occurred!
See also "/home/gaspar/git/SoapySDR/build/CMakeFiles/CMakeOutput.log".
See also "/home/gaspar/git/SoapySDR/build/CMakeFiles/CMakeError.log".
```
Workaround:
```bash
cmake -DCMAKE_INSTALL_PREFIX=$VIRTUAL_ENV -DPYTHON_INSTALL_DIR=(python -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())") ..
```

guruofquality, 2019-09-20:
seems like others have seen this, https://github.com/pypa/virtualenv/issues/737 and so its a general issue with this software for other projects that also use this API for install: https://github.com/pypa/virtualenv/issues/355

gasparka, 2019-09-20:
Maybe the ```site.getsitepackages()``` call can be replaced with ```python -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())"```?

guruofquality, 2019-09-20:
then a bunch of other people will have problems -- hence the need for getsitepackages.

Take a look: https://github.com/pothosware/SoapySDR/blob/master/python/get_python_lib.py

Its already using get_python_lib, the script would need a workaround like a try/except around getsitepackages since it doesnt exist in this case. But it still seems wrong if its a problem with virtualenv's site.py in the first place

closed 2019-09-25 by guruofquality