pothosware/SoapySDR #242: Low TX power even at max gain ?

OhSoGood, 2019-09-24:
Hi,
I get low TX power whatever the gain I choose using soapy, on my limesdr usb. I get -33dBm and wonder how to reach the max level (0dBm at 2.4GHz) that LimeMicro announces on its Myriadrf. Can anyone help?

To verify the TX level, I connect my LimeSDR to test via a direct cable to another SDR (e.g. a 2nd lime) or a spectrum analyzer. Even when TX gain is close to its max (here 60), and whatever the signal I send around 2.4GHz, I measure an output of -31 to -33dBm. Though, when the smart guys from LimeMicro use their own raw tool LimeSuite, they get around 0dBm... I'm not the only one facing this (see https://discourse.myriadrf.org/t/output-power-limesdr/2207/14 ) but I'm wondering whether around 0dBm is achievable in soapy, and who to do it.

zuckschwerdt, 2019-09-24:
Did you select the right antenna output? Happened to me more than once.

OhSoGood, 2019-09-24:
You mean the right channel and the right band (ie RX1_H for the Lime)? If so, yes, pitifully.

zuckschwerdt, 2019-09-24:
Yes. I test sending with tx_sdr from [tx_tools](https://github.com/zuckschwerdt/tx_tools/), it works well on my LimeSDR mini. Didn't test a LimeSDR USB recently though.

OhSoGood, 2019-09-30:
@zuckschwerdt : any news about it? 
I measure the loss due to the cable (-1.5dB), so measured output should be offset by as much. Still, that doesn't lead me to 0dB like LimeMicro is able to do with the raw LimeSDR API.

zuckschwerdt, 2019-09-30:
I only did some cursory tests. I need to setup to really measure the output. To clarify, the full range of dB settings works for you (showing the expected increments) but the max settings is below the advertised dBm? Which dB range is reported (`SoapySDRUtil --probe`) for your LimeSDR USB?

OhSoGood, 2019-09-30:
Yes, that's it.
- Max range is as expected: [0, 64] globally, in which there is [0, 52] for PAD and [-12, 12] for IAMP.
- I use a gain of 52 so that PAD=52 and IAMP=0. Beyond 52, output level is no more linear with gain (saturation occurs).
- At gain 52, the output I measure is around -33dBm.

closed 2022-11-24 by ncorgan