pothosware/SoapySDR #249: SdrGlut has some problems with SoapySDR

righthalfplane, 2019-12-10:
SdrGlut has some problems with SoapySDR.

https://github.com/righthalfplane/SdrGlut

First the good news  -

SdrGlut works well with a Lime Mini, HackRF One, SDRPlay and NetSDR using SoapySDR on - 

MacOS Sierra

MacOS High Sierra

MacOS Mojave

Ubuntu 16.04 - after some fiddling around

Raspberry Pi 4 - normal edition of Raspbian 

Raspberry Pi 4  - the SdrPlay of Raspbian 

On the Raspberry Pi 4 using the Lime Mini or the HacKRF One, SdrGlut managed 20 Msps where CubicSDR could only do less than 10 Msps. 

https://www.youtube.com/watch?v=C18bGxbXNsA

The problems -

On Windows 7 , trying to use the Lime Mini usually crashes CubicSDR and SdrGlut usually runs, but will only does 1 Msps with out dropping samples. The HacKRF One runs nicely at its maximum 20 Msps and SdrPlay does its 10 Msps.

Ubuntu 18.04, CubicSDR finds the HackRF One, but SdrGlut does not find it - that is very strange since the are both using SoapySDR to search for it.

guruofquality, 2019-12-15:
> Ubuntu 18.04, CubicSDR finds the HackRF One, but SdrGlut does not find it - that is very strange since the are both using SoapySDR to search for it.

Usually when that happens its due to having more than one version of SoapySDR installed, so sdrglut looking at one SoapySDR.so, and Cubic looking at another. It can happen when stuff from both the package manager and development files get installed at the same time. I would check the usual paths /usr/local/lib/libSoapySDR.so* and /usr/lib/x86_64-linux-gnu/libSoapySDR*

righthalfplane, 2019-12-16:
Yes, that was it - There were actually three versions of SoapySDR installed. Ubuntu had installed two versions,0.6 and 0.7 and I had installed 0.8. For some reason, when I built SdrGlut it grabbed the 0.7 version that had no modules installed instead of the 0.8 that I was installing. CubicSDR was using the 0.6 version, which has some modules installed.

closed 2022-11-24 by ncorgan