pothosware/SoapySDR #250: Can't compile python package with python 3.8 and swig 4.0.1

Dantali0n, 2019-12-29:
When I try to compile SoapySDR 0.7.1 the compilation of python bindings / package explodes in a huge list of undefined reference errors:

https://paste.ubuntu.com/p/W4FpKcsmzD/

Some other perhaps relevant info:
```
$ neofetch
██████████████████  ████████   corne@Dantali0n 
██████████████████  ████████   --------------- 
██████████████████  ████████   OS: Manjaro Linux x86_64 
██████████████████  ████████   Host: P64_HJ,HK1 
████████            ████████   Kernel: 5.4.6-2-MANJARO 
████████  ████████  ████████   Uptime: 3 hours, 28 mins 
████████  ████████  ████████   Packages: 2192 (pacman) 
████████  ████████  ████████   Shell: bash 5.0.11 
████████  ████████  ████████   Resolution: 1920x1080 
████████  ████████  ████████   WM: i3 
████████  ████████  ████████   Theme: Adwaita [GTK3] 
████████  ████████  ████████   Icons: Adwaita [GTK3] 
████████  ████████  ████████   Terminal: xfce4-terminal 
████████  ████████  ████████   Terminal Font: Monospace 12 
                               CPU: Intel i7-7700HQ (8) @ 2.800GHz 
                               GPU: Intel HD Graphics 630 
                               GPU: NVIDIA GeForce GTX 1050 Ti Mobile 
                               Memory: 2400MiB / 32004MiB
```

Dantali0n, 2019-12-29:
Seems fixed already on master~

closed 2019-12-29 by Dantali0n

stef, 2020-01-09:
can you link to the commit so others who run into this don't have to dig it out until there is a release which includes that patch?

stef, 2020-01-09:
is it this patch? https://github.com/pothosware/SoapySDR/commit/4dcd956d1a79fbb346303e786ac2d9448f191ba3

Dantali0n, 2020-01-10:
I have no clue which patch fixed my problem and no time to find out. I switched from 0.7.1 to master and that fixed it. Your patch seems like a likely guess.

guruofquality, 2020-01-12:
I tagged a release with this fix in it

stef, 2020-01-12:
woooho, thanks very much! \o/