pothosware/SoapySDR #252: Device gets disconnected after RX

kehak, 2020-01-09:
Hello,

I'm trying SoapySDR with an Airspy R2 through its Python interface.

The Airspy is properly connected:

    $ SoapySDRUtil --probe="driver=airspy"
    ######################################################
    ##     Soapy SDR -- the SDR abstraction library     ##
    ######################################################
    
    Probe device driver=airspy
    
    ----------------------------------------------------
    -- Device identification
    ----------------------------------------------------
      driver=Airspy
      hardware=Airspy
      device_id=0
      origin=https://github.com/pothosware/SoapyAirspy
    
    ----------------------------------------------------
    -- Peripheral summary
    ----------------------------------------------------
      Channels: 1 Rx, 0 Tx
      Timestamps: NO
      Other Settings:
         * Bias tee - Enable the 4.5v DC Bias tee to power SpyVerter / LNA / etc. via antenna     connection.
           [key=biastee, default=false, type=bool]
         * Bit Pack - Enable packing 4 12-bit samples into 3 16-bit words for 25% less USB trafic.
           [key=bitpack, default=false, type=bool]
    
    ----------------------------------------------------
    -- RX Channel 0
    ----------------------------------------------------
      Full-duplex: NO
      Supports AGC: YES
      Stream formats: CS16, CF32
      Native format: CS16 [full-scale=32767]
      Antennas: RX
      Full gain range: [0, 45] dB
        LNA gain range: [0, 15] dB
        MIX gain range: [0, 15] dB
        VGA gain range: [0, 15] dB
      Full freq range: [24, 1800] MHz
        RF freq range: [24, 1800] MHz
      Sample rates: 10, 2.5 MSps

Now if I try to run the sample code available here under "Basic example":
https://github.com/pothosware/SoapySDR/wiki/PythonSupport

It works great also and I get some samples. However, I cannot re-run it because the device gets unbound, as shown with  `udevadm monitor` :

    KERNEL[83347.594892] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0 (usb)
    UDEV  [83347.598439] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0 (usb)
    KERNEL[83347.601655] unbind   /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0 (usb)
    UDEV  [83347.602938] unbind   /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0 (usb)
    KERNEL[83347.607550] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0 (usb)
    UDEV  [83347.610574] bind     /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0 (usb)
    KERNEL[83347.614322] unbind   /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0 (usb)
    UDEV  [83347.616109] unbind   /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0 (usb)

    $ SoapySDRUtil --probe="driver=airspy"
    ######################################################
    ##     Soapy SDR -- the SDR abstraction library     ##
    ######################################################
    
    Probe device driver=airspy
    Error probing device: device_id missing.

Sometimes the device doesn't get disconnected, but every sample received afterward is NULL.
The solution is to unplug and plug back the device, it then resumes working.

Is this a known issue? How can I help troubleshoot this?

guruofquality, 2020-01-12:
Its not a known issue. Just making calls into libairspy shouldnt disconnect the device from the USB bus as far as I know. I can only imagine driver issues or flakey usb cables, power supplies.

kehak, 2020-01-15:
Thank you for your answer. I can rule out dubious hardware as I can replicate the problem on two machines. This could be a driver issue though. I notice that after acquiring some samples with Soapy, the Airspy appears as disconnected but runs warm as it if it was still streaming and its DSP working.

I'm not sure how to move forward with this issue. This is an Airspy R2 with firmware 1.0.0-rc10 (latest).
I guess I will call libairspy directly from Python as a workaround, since it works fine with libairspy and airspy_rx.

closed 2022-11-24 by ncorgan