pothosware/SoapySDR #253: modules0.8 (missing) macOS 10.15

ah1102, 2020-03-06:
I'm trying to compile from this instruction https://github.com/cjcliffe/CubicSDR/wiki/Build-OSX

```
ccliffe $ cd ~ / Dev
Dev $ git clone https://github.com/pothosware/SoapySDR.git
Dev $ cd SoapySDR
SoapySDR $ mkdir build
SoapySDR $ cd build
build $ cmake .. -DCMAKE_BUILD_TYPE = Release
build $ make
build $ sudo make install
build $ SoapySDRUtil --info
```

I tried the original author https://github.com/pothosware/SoapySDR/wiki/BuildGuide#osx The result is the same

```
Apple@iMac-Apple build % soapysdrutil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-gf722f9ce
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8 (missing)
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

Here is the build output

```
Apple@iMac-Apple build % sudo cmake .. -DCMAKE_BUILD_TYPE=Release
-- 
-- #############################################
-- ## Begin configuration for Python support...
-- #############################################
-- Enabling optional Python bindings if possible...
-- SWIG_FOUND: TRUE - 3.0.12
-- PYTHONINTERP_FOUND: TRUE - 2.7.16
-- PYTHON_EXECUTABLE: /usr/bin/python
-- PYTHON_INSTALL_DIR: ${prefix}/lib/python2.7/site-packages
-- PYTHONLIBS_FOUND: TRUE - 2.7.16
-- PYTHON_INCLUDE_DIRS: iwithsysroot;/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7;-iwithsysroot;/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7
-- PYTHON_LIBRARIES: -L/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/config -lpython2.7 -ldl -framework CoreFoundation
-- CMAKE_SWIG_FLAGS=-c++;-threads;-I/Users/Apple/Dev/SoapySDR/include
CMake Warning (dev) at /usr/local/Cellar/cmake/3.16.5/share/cmake/Modules/UseSWIG.cmake:607 (message):
  Policy CMP0078 is not set: UseSWIG generates standard target names.  Run
  "cmake --help-policy CMP0078" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

Call Stack (most recent call first):
  python/CMakeLists.txt:155 (SWIG_ADD_LIBRARY)
This warning is for project developers.  Use -Wno-dev to suppress it.

[... 107 lines cut ...]
-- Up-to-date: /usr/local/include/SoapySDR/Registry.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Errors.h
-- Up-to-date: /usr/local/include/SoapySDR/Config.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Version.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Modules.h
-- Up-to-date: /usr/local/include/SoapySDR/Constants.h
-- Up-to-date: /usr/local/include/SoapySDR/Device.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Modules.hpp
-- Up-to-date: /usr/local/include/SoapySDR/ConverterRegistry.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Logger.h
-- Up-to-date: /usr/local/include/SoapySDR/Types.hpp
-- Up-to-date: /usr/local/include/SoapySDR/Formats.hpp
-- Up-to-date: /usr/local/lib/libSoapySDR.0.8.0.dylib
-- Up-to-date: /usr/local/lib/libSoapySDR.0.8.dylib
-- Up-to-date: /usr/local/lib/libSoapySDR.dylib
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRExport.cmake
-- Up-to-date: /usr/local/share/cmake/SoapySDR/SoapySDRExport-release.cmake
-- Up-to-date: /usr/local/lib/pkgconfig/SoapySDR.pc
-- Installing: /usr/local/bin/SoapySDRUtil
-- Up-to-date: /usr/local/share/man/man1/SoapySDRUtil.1
-- Installing: /usr/local/lib/python2.7/site-packages/_SoapySDR.so
-- Up-to-date: /usr/local/lib/python2.7/site-packages/SoapySDR.py
-- Installing: /usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/_SoapySDR.so
-- Up-to-date: /usr/local/Cellar/python/3.7.6_1/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/SoapySDR.py
```

zuckschwerdt, 2020-03-06:
That's expected, you don't have any modules installed. It's optional, but you likely want some modules like [SoapyRTLSDR](https://github.com/pothosware/SoapyRTLSDR).
Perhaps try the CubicSDR binary releases, those include SoapySDR and modules. Or install SoapySDR and modules from MacPorts or Brew.

closed 2020-03-06 by zuckschwerdt

ah1102, 2020-03-06:
I do not understand. I need to compile the latest version of Cubicsdr from source. According to the instructions for this, you need to collect Soapysdr from the source. I stopped at this step, since soapysdrutil --info on the path /usr/local/lib/SoapySDR/modules0.8 (missing) cannot find the module, since it is not there.

You can see in the logs that nothing is created on the path / usr / local / lib / SoapySDR. I wonder why so?

When executing cmake in the log, there are some caveats about Python. Do you need it in osx? The instructions say nothing about this.

SDRplay, 2020-03-06:
I think @zuckschwerdt point is that the /usr/local/lib/SoapySDR and modules0.8 directories are not created until you build and install a library (e.g. SoapySDRPlay or SoapyRemote, etc.) - I can't remember from memory the sequence of events, but I know that after installing the libraries you definitely should have those directories - so that's worth you doing before going on to CubicSDR anyway.

ah1102, 2020-03-06:
SDRplay thanks for helping! I want to tell you that if you are a representative of sdrplay.com, then on your site Cubicsdr 0.2.6 release swears at the lack of soapysdr. Why do you think I started trying to collect 0.2.6 myself?)

[image]

SDRplay, 2020-03-06:
We include SoapySDR and SoapySDRPlay in the API installer (which is a requirement for SDRplay products) - so just 2 downloads/installs. If you have a different SDR device, then you'll have to consult with the manufacturer as to what you need I'm afraid.

ah1102, 2020-03-06:
I do not connect any device at first. First I download 0.2.5 and when I start the program does not swear and expects the device to connect. Then I download 0.2.6 and when the program starts, a warning window appears that soapysdr are not installed.

SDRplay, 2020-03-06:
I'm confused - why would you build/install SoapySDR but not any of its libraries? I think this will be a non-issue once you've got SoapySDR libaries installed.

ah1102, 2020-03-06:
Does soapysdr install inside .app or should it be installed on the user machine? I’ll try to install soapySDRrtl now, but according to cubic’s wiki, soapysdr is needed first, and the rest of soapy is optional. Therefore, I am trying to collect it.

Here is what gives 0.2.5
[image]

 Here is what gives 0.2.6
[image]

Moreover, on my computer it is:

```
Apple@iMac-Apple build % soapysdrutil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-gf722f9ce
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8 (missing)
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

SDRplay, 2020-03-06:
IF both of those CubicSDR applications are using the SAME SoapySDR, then clearly the issue is with CubicSDR, correct?

If so, then you'll probably want to raise an issue on the CubicSDR github. As I said, I've only got experience with installs using SoapySDRPlay, so probably not the best person to comment on what CubicSDR is doing in general.

You could always create those directories to see what SoapySDRUtil then says - sorry I don't seem to be helping you.

ah1102, 2020-03-06:
Friends, no one answered the question to me whether I need to install Python to build soapysdr under macos? The wiki soapysdr manual says nothing about this. But for linux and windows it says that python should be.

guruofquality, 2020-03-07:
Python bindings are optional and not required for building.