pothosware/SoapySDR #254: Build instructions missing

fyngyrz, 2020-04-05:
On [this SoapySdr github page](https://github.com/pothosware/SoapySDR/wiki/BuildGuide), the OSX  instructions say to use homebrew/porthos \(impossible &mdash; can't install homebrew as it has system conflicts in my setup\), or "follow the build instructions below."

However, there are no OSX build instructions, and looking inside the cloned repository, there is no makefile, or configure, or XCode project, or Qt project (I have Qt installed and [my software](is built using Qt)[https://fyngyrz.com/?page_id=995].

I would very much appreciate a step-by-step set of instructions for building SoapySdr under OSX.

Thank you.

guruofquality, 2020-04-05:
The normal unix instructions should do, just make sure you have cmake and make installed first: https://github.com/pothosware/SoapySDR/wiki/BuildGuide#unix-instructions

fyngyrz, 2020-04-05:
Does the following indicate a successful install?

```
Lib Version: v0.8.0-gf722f9ce
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8 (missing)
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

SDRplay, 2020-04-05:
yes, so far so good :-)

fyngyrz, 2020-04-05:
Great.

I've edited the OSX [install instructions](https://github.com/pothosware/SoapySDR/wiki/BuildGuide) a wee bit to include the above info. Hopefully that's okay.

nickoe, 2020-12-28:
@guruofquality ore @fyngyrz I think this can be closed

closed 2020-12-28 by guruofquality