pothosware/SoapySDR #255: SoapySDR - can I build this for static linking into my application?

fyngyrz, 2020-04-06:
**Qt** is acting cranky (okay, more like bewildered) about linking .dylibs, and aside from that, I'd just as soon not have yet more files cluttering up the various application installs for distribution to Windows, OSX, and Linux anyway.

How (or can) I go about creating a statically linkable <tt>libSoapySDR</tt> for my build?

fyngyrz, 2020-04-06:
more... okay, I found homebrew mechanisms to install modules - which apparently actually *builds* them in place, which is outright horrifying, but regardless, homebrew is regrettably OS-version-specific and declines to operate properly unless it has versions of the OS and XCode that Apple won't even allow to run on my hardware, so no module builds for me.

I must be approaching this all wrong? I hope?

fyngyrz, 2020-04-06:
FWIW, I'm running with the dylibs now... but without modules, I'm at an impasse.

fyngyrz, 2020-04-06:
I guess this is off topic for my original question about static builds; I'll open a new one...

zuckschwerdt, 2020-04-06:
Generally you should be fine with dynamic linking when developing your app. Pull SoapySDR and the modules from Homebrew, MacPorts or just build and install (cmake). You'll likely need some hardware specific stuff (libmirsdrapi-rsp.dylib for SDRplay). If you want to distribute a complete app you'll need to package the libs or instruct your users to install them.
TL;DR if Homebrew seems to intrusive (it's supposed to link to system libs) use MacPorts -- it's cleanly contained with little system deps.

fyngyrz, 2020-04-06:
> Pull SoapySDR and the modules from Homebrew

Homebrew is pretty broken under OSX 10.12.6, and/or the modules are, resulting in failure to build some modules (e.g. AirSpyHF won't build.)

> if Homebrew seems to\[o\] intrusive

It's not that it's intrusive, it's that it doesn't actually succeed in building modules. Or the modules themselves are too OS-dependent. I guess it doesn't matter which; if they won't build, they won't build.

> If you want to distribute a complete app you'll need to package the libs or instruct your users to install them.

Hm. So Soapy doesn't actually encapsulate SDR support, but depends upon further external resources. I was under a different impression; I'll have to think about this, it changes the nature of the project a bit, or at least, what modules I'm willing to include with the server. I have the RTL SDR module loading, so I'm at least off to some kind of a start. :)

closed 2022-03-25 by zuckschwerdt