pothosware/SoapySDR #256: Where can I get Soapy modules, already built, for my users?

fyngyrz, 2020-04-06:
I'm working on a server that I hope to provide my users with a Soapy-based interface between my [SDR application](https://fyngyrz.com/?p=915) and various Soapy supported SDRs. I have the build working, and it discovers the RTL module, which I managed to get built. I can use that to complete the server, presuming the following issue can be addressed:

The problem I'm having is, *SDR module availability.* What I've found so far requires being built via `homebrew`, and `homebrew` rarely runs properly / to completion on my 12/24-core, 64GB OSX 10.12.6 machine, as Apple declines to let me upgrade the OS any further, and the `homebrew` people have, for whatever reason, imposed some dependencies on their system for (a) later version(s) of MacOS. In this case, `homebrew` built the RTL module (so at least I can test with that), but it refuses to build the airspyhf module, for instance.

So: Is there a place where pre-built drivers for various SDRs are available, that can simply be installed in the proper folder and then are usable?

I'm particularly interested in functional SdrPlay and AirSpy driver binaries.

zuckschwerdt, 2020-04-07:
MacPorts runs fine on Mac OS X 10.12 Sierra. They have the modules prebuilt also, e.g. http://packages.macports.org/SoapySDRPlay/
Maybe try: `port install SoapySDR SoapyAirspy SoapyAirspyHF SoapyAudio SoapyBladeRF SoapyHackRF SoapyPlutoSDR SoapyRedPitaya SoapyRemote SoapyRTLSDR SoapySDRPlay SoapyUHD`
Using CMake to build the modules is working cleanly too. You would need to collect all the interesting modules and the base libraries that are wrapped.

Caveat: SoapySDR has a module lookup path and that would be specifc to the installation method, e.g. /usr/local/... when building yourself or /opt/local/... with MacPorts. You would likely want to customize that path to point to some resource folder in your application.

fyngyrz, 2020-04-07:
> They have the modules prebuilt also

Thank you so much for the link to the built modules. 

So far, I've used MacPorts to install them; and a lot was installed, including a bunch of device-specific libraries. However, my application only finds the RTL SDR; and looking in `/opt/local/` I didn't find anything but the `.tgz` files; I have both the SdrPlay and the AirSpyHF attached to the system as well as the RTL stick, so I would have expected them to be found by my app if the install had worked.

Next, I'll try downloading the modules directly and putting them wherever the RTL module ended up. Now all I have to do is find that - at least I know it's around and accessible, as my app does find it via SoapySDR.

So much to learn...

fyngyrz, 2020-04-07:
Further... I found the modules, and I found all manner of libraries and commands scattered around (airspy-this-and-that, etc.), and I copied them from the macports locations to the same location as my rtlsdr lib and module files... still no joy

fyngyrz, 2020-04-07:
Can you tell me if users install SoapySDR on their own, will it normally install everything they need? To put it another way, if I can get the RTL support, when selected, fully up and working with all the various antenna, gain, ranges, sample rates and so on, is it reasonable to expect SoapySDR will take care of the rest for them? I'm really worried about this, because the install here is being so uncooperative.

I'm really concerned with how difficult this is to get going on my end. There's no way I could expose my users (well over 20,000 of them) to this kind of thing &mdash; they'll outright _bury_ me in requests for support.

fyngyrz, 2020-04-07:
With the 0.8.0 library, I get this...

[image]

...Which, when I select the RTL SDR, allows my app to set it up and display whatever options there are...

[image]

..when I link with the 0.7.2 library, I get this...

[image]

...which at first thought "oh, it's giving me audio sources to do IQ streaming from", but then (a) the RTL SDR is now missing, which might be because _that_ module is the wrong ABI now (I'll chase that next), but worse, (b), selecting one of those sources blows up the sdr selection code. Going to see if I can debug what's happening there, presuming that I should even be getting those audio devices...

fyngyrz, 2020-04-07:
ok, utilizing the 0.7.2 Soapy lib,  I'm doing the following once the item is selected (when picking Apple built-in line input):

```
str_list = sdr->listGains( SOAPY_SDR_RX, 0);
if (str_list.size() > 0)
```

`str_list.size()` is returning `6148914691236517205` (lol) which causes additem loop to crash when I try to load the drop-down with the details in the list. So I either shouldn't be calling this with audio devices (how would I know that? There's nothing in the [sample code](https://github.com/pothosware/SoapySDR/wiki/Cpp_API_Example) that indicates this) or the library is broken, either in that it's even showing me these items, or in that it's giving me an insane return when it really should give me zero if there are no selections to be made.

Do you have any tips for me here?

fyngyrz, 2020-04-07:
I wrapped the above with...

```
if (sdr->hasGainMode(SOAPY_SDR_RX, 0))
{
}
```

...thinking that it might catch the issue, but no, it returns `true` and the returned length of the list continues to be absurd. More anon...

fyngyrz, 2020-04-07:
yeah... if I remove the requests / enumerations for gains, frequency range, and samplerate range, selecting these audio items doesn't (at least) crash. In all cases, I'm getting invalid results.

SDRplay, 2020-04-07:
I don't know if it will help you, but when you install the API from our website, we also install SoapySDR and SoapySDRPlay ready to go. - https://www.sdrplay.com/downloads

fyngyrz, 2020-04-07:
okay, I'm stuck now. Will wait on illumination from someone expert in these matters before I put any more time into this.

fyngyrz, 2020-04-07:
> I don't know if it will help you, but when you install the API from our website, we also install SoapySDR and SoapySDRPlay ready to go. - https://www.sdrplay.com/downloads

Hmm. Does your install set up all of the modules, or just your own?

fyngyrz, 2020-04-07:
I've parsed the iteration of the device to see if the driver tag is "audio" and if so, I'm not asking for those three settings; that, at least, has made it conditionally skip the broken library calls.

Now if I can get the RTL driver up and running with the 0.7.2 version... have to find the 0.7.2 RTL driver...

SDRplay, 2020-04-07:
We only provide and support the libraries for our own devices.

guruofquality, 2020-04-07:
@fyngyrz you may want to read this: https://github.com/pothosware/SoapySDR/wiki/ConfigGuide#avoid-simultaneous-installs You need to be really attentive when it comes to dev environments, search paths, execution paths. I think this at least explains what you are seeing.

fyngyrz, 2020-04-07:
> We only provide and support the libraries for our own devices.

Okay, thank you.

Can you tell me what version your api bundle would install? 0.7.2, 0.8.0, or something else?

I have an RTL driver for 0.8.0 that works, and audio drivers that _might_ work for 0.7.2; so I just want to know what I would be getting into with a _fourth_ install of SoapySDR before I jump. :)

fyngyrz, 2020-04-07:
> You need to be really attentive when it comes to dev environments, search paths, execution paths. I think this at least explains what you are seeing.

@guruofquality :: I understand that.

Unfortunately, when the 0.8.0 dev version didn't install the available modules, and the porthos homebrew mechanism  failed mid-install and left me with who-knows what, and macports installed 0.7.2 &mdash; that's a mess not of my own making, but now, there it is.

I can switch between them &mdash; the respective dylibs look at different filesystem locations to find modules and libraries &mdash; by switching which .dylib  I link my app to. I want to get things settled down to one or the other, but I have yet to find a consistently working set of modules/libraries that matches either the macport 0.7.2 dylib or the dev 0.8.0 dylib.

The development environment should provide everything needed all at one fully compatible revision level and install it all properly; unfortunately, that is not the case. And then there are the [bugs in the SoapySDR device enumeration calls](https://github.com/pothosware/SoapySDR/issues/257) complicating matters.

Another possibility &mdash; yet another install /sobs/ &mdash; lurks with the SdrPlay folk's downloadable API install, but I'm waiting to hear what version _that_ installs before I jump into potentially deeper waters.

SDRplay, 2020-04-07:
0.8

fyngyrz, 2020-04-07:
@SDRplay :: Running OSX 10.12.6 here. Should I install V2.13 or V3.07 ?

The unit has the following on the back...

**_SDRPlay_** MODEL **_RSP_**
**Wideband USB Radio Receiver**
\[FCC\] \[CE\] \[no-trash\] \[✔️RoHS\]
**SN** - `RSP#121115#10#425`

...black plastic case, one antenna connector, one USB Type-B connector.

SDRplay, 2020-04-08:
API 2.13 version is almost 2 years old now so that will be SoapySDR 0.7

API 3.x is currently in beta testing. This is SoapySDR 0.8

We've had reports that on some versions of MacOS the 3.x API has some issues which we are looking at. I'm running it right now on El Capitan without a problem.

Hope that helps.

Andy

fyngyrz, 2020-04-08:
@SDRplay :: **_Thank you_**. I downloaded your install, backed up the 0.8.0 level RTL stick module, uninstalled the macports and homebrew versions, installed yours, copied back the RTL stick module, and what do you know... it all works. :)

closed 2022-03-25 by zuckschwerdt