pothosware/SoapySDR #257: Bug report: list requests return insane values

fyngyrz, 2020-04-07:
When the sdr device provides an audio device driver...

```
str_list = sdr->listGains( SOAPY_SDR_RX, 0);
howMany = str_list.size();
```

...`str_list.size()` returns an invalid (very large) value to `howMany`

If there are no gain items to be had,`str_list.size()` should return **zero.**

This is also the case for `sdr->getFrequencyRange` and `sdr->getSampleRateRange` so three bugs, really.

guruofquality, 2020-04-07:
Its probably an ABI error. Mismatch in development source vs runtime library. Or possibly even macos libstdc++ libs which I have seen on clang compiles in the past if the flags werent set the same as the library being used.

I cant tell if this is something between the application/test case here and the SoapySDR library or between the SoapySDR library and one of its modules. In your test case above, if you see the same problem with the null device, then its probably an issue compiling the application. `auto device = SoapySDR::Device::make("driver=null");`

fyngyrz, 2020-04-07:
> Its probably an ABI error. Mismatch in development source vs runtime library.

I can see how that could happen, as it occurs with the 0.7.2 dylib vs. my app, which was built against 0.8.0 &mdash; so I guess this is just an unfortunately fragility consequent of the churn between 0.7.2 and 0.8.0.

The dev stuff I have is 0.8.0; so what I need, then, are 0.8.0 modules, prebuilt, so as to get past this properly. The SdrPlay folk apparently have prebuilt 0.8.0 stuff for their SDR, and as soon as I hear back from them as to which API to download, I'll (probably) have another working module. Two is good for testing; I can switch back and forth and make sure they both set up correctly, etc. Here's hoping, anyway.

Feel free to close this &mdash; if these calls are working for you, then it's on my end, and probably right where you're pointing, too. :)

closed 2020-11-01 by ncorgan