pothosware/SoapySDR #259: CMake 3.1.0 or higher is required.

scaprile, 2020-04-17:
As there is no binary for CentOS 7 (afaik) I tried to build my own.
There are almost no dependencies, but one of the undocumented dependencies is the version of cmake needed to build.
CMake 3.1.0 or higher is required.  You are running version 2.8.12.2
Not able to build on RHEL7 derivatives (afaik)

zuckschwerdt, 2020-04-18:
CMake 3 was released in 2014, I was going to suggested updating base packages. It seems that some distros really still use 2.8.12 though. Here is [a list of RHEL, CentOS and Fedora](https://rpms.remirepo.net/rpmphp/zoom.php?rpm=cmake). Are backports available? You can try to remove the requirement check, but expect to rewrite some CMake commands.

scaprile, 2020-04-18:
There seems to be a cmake3 package and doesn't look interfering so far. I will try and let you know. Rewriting CMake commands is beyond my skill level.
Thanks

scaprile, 2020-04-20:
Compiles OK using cmake3 in the command line.

closed 2020-04-20 by scaprile